hku-mars/FAST_LIO #1: Pose largely drift

YangSiri, 2020-12-12:
Hi, @XW-HKU 

I compiled Fast-LIO in ubuntu16.04 and I tried to test it with Avia. But the result may be totally wrong, I'm curious about what I may have done to make this kind of strange phenomenon?

[image]

The pose may drift even thought it is still.

Looking forward to your help  ;>

XW-HKU, 2020-12-12:
1. Which launch file you use?
2. Please make sure the version of pcl and ubuntu meet the requirements writen in the project readme.

YangSiri, 2020-12-12:
> roslaunch fast_lio mapping_avia.launch
> roslaunch livox_ros_driver livox_lidar_msg.launch

I try to test it on ubuntu16.04 with default pcl. I see the system runs normally without any errors but the result is strange. 
I was running real time. I will record some bag and send to you later. Thanks for your answer!

YangSiri, 2020-12-12:
Hi, @XW-HKU 

The _2020-12-12-18-10-36_aviaTest.bag_ has been sent to your email.

XW-HKU, 2020-12-12:
Here are three thing may help you:
1. The Avia LiDAR cannot produce good measurement inside the 10m distance between surroundings and liDAR
2. FAST_LIO only support ubuntu >= 18.04 and pcl >= 1.8, please look at the homepage of this github repo.
3. In the bag you sent me, FAST LIO can work well at the first 11s, but after 11s, the lidar is staying before a wall closer than 2m, which will make the liDAR produce meaningless points.

YangSiri, 2020-12-12:
Thanks, @XW-HKU 

You mean the bag works normally in your environment which is 18.04 except the situation you mentioned?

YangSiri, 2020-12-12:
Thanks, @XW-HKU

closed 2020-12-12 by YangSiri