hku-mars/FAST_LIO #5: #include <fast_lio/States.h> not present

fperasso, 2021-02-16:
I would like to buy an AVIA, but I tried to install your FAST_LIO package first, but once the ros_melodic - PCL OPENCV and eigen packages are installed, when I download your ghitub it gives me a privacy error I had to comment the line "### cmake_policy (SET CMP0074 NEW) "otherwise the compilation does not proceed and now I have the problem that gives me an error because you have not inserted in the [include] folder a library #include <fast_lio / States.h>, now you can correct the problem and insert all the right libraries? . Another thing seems that the vs ghitub must reside inside the src of the space where livox_ros_driver also resides otherwise if you put it in a different space at the time of catkin_make it does not find the ros_drive directory and the compilation stops. Can you explain better and be more precise in all the steps otherwise we will not be able to proceed, unfortunately we are not as expert programmers as you are? Thank you
this is the terminal error:
**[ 64%] Building CXX object FAST_LIO/CMakeFiles/loam_laserMapping.dir/src/laserMapping.cpp.o
In file included from /home/livox2/ws_livox/src/FAST_LIO/src/laserMapping.cpp:47:0:
/home/livox2/ws_livox/src/FAST_LIO/include/common_lib.h:8:10: fatal error: fast_lio/States.h: File o directory non esistente
 #include <fast_lio/States.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
FAST_LIO/CMakeFiles/loam_laserMapping.dir/build.make:62: recipe for target 'FAST_LIO/CMakeFiles/loam_laserMapping.dir/src/laserMapping.cpp.o' failed
make[2]: *** [FAST_LIO/CMakeFiles/loam_laserMapping.dir/src/laserMapping.cpp.o] Error 1
CMakeFiles/Makefile2:3616: recipe for target 'FAST_LIO/CMakeFiles/loam_laserMapping.dir/all' failed
make[1]: *** [FAST_LIO/CMakeFiles/loam_laserMapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
livox2@livox2-VirtualBox:~/ws_livox$**

xuankuzcr, 2021-02-17:
You can try to repeat "CMake" several times.

fperasso, 2021-02-17:
sorry but the fast_lio / states library where is it? it shouldn't be in the include directory of your package, otherwise how can it compile? Where should it be?

XW-HKU, 2021-02-18:
> sorry but the fast_lio / states library where is it? it shouldn't be in the include directory of your package, otherwise how can it compile? Where should it be?

thanks for your suggestion，the states.h is automatically generated by the .msg file in msg folder. This is a common ros technique

fperasso, 2021-02-19:
Hello I would like to ask you another question, we are in the process of buying an Avia sensor to experiment with your FAST_LIO package, I managed to install your github on a virtual machine with ubuntu 18.04 LTS and also on jetson TX2 Nvidia which is a system recommended by DJI for livox systems, the github is all compiled. Do you think the system also works on tx2? I would like to avoid buying 2c manifold too which costs a lot and is not available here in Italy. What do you advise me to do, have you done any tests on a computer with jetson TX2? My team and I would like to put your system on a dji drone to carry out tests on hollow mining sites in very large spaces even 400mt to verify the accuracy of the clouds, then we will compare your system with a TOPCON laser scanner to analyze the differences. We would be interested in integrating precise GNSS positioning to your system to georeference and orient the cloud topographically, within 2cm of precision. Are you developing something to merge IMU and PointClaud data with Rinex data? There are very cheap UBLOX modules such as ublox FP9 that can manage dual band L1 L2 gnss signals which, if doubled, can create the baseline for orientation in space topographically. It would be interesting to combine your Fast_Lio package with a LOG system with satellite rinex data that aligns the cloud with centimeter precision which could then in post processing reach below the centimeter and color the clouds with a Sony a6000 camera. Hello from Italy.

XW-HKU, 2021-02-19:
> Hello I would like to ask you another question, we are in the process of buying an Avia sensor to experiment with your FAST_LIO package, I managed to install your github on a virtual machine with [...]

Thanks for your interests in our works.

Here are something you may be interested in.

1. We will launch Fast-LIO 2.0 in coming March. The Fast-LIO 2.0 will support the real-time implementation in embedded systems including TX2, Raspberry Pi 4 and other ARM-based platforms. But current version of Fast-LIO cannot guarantee the real-time performance in TX2, especially for the Large-scale space. If you are still interested in implementing Fast-lio, please wait for the Fast-LIO 2.0.

2. FAST-LIO is focused on the fusion of IMU and LiDAR, not considering other sensor like GPS or Vision. But our MaRS Lab will propose some research projects that fusion other sensors, in a few months. So stay tuned!

mhkabir, 2021-03-24:
@XW-HKU is FAST_LIO 2.0 the same as r2live?

XW-HKU, 2021-07-07:
> @XW-HKU is FAST_LIO 2.0 the same as r2live?

r2live use fast-lio as lidar-inertial front-end.

closed 2021-07-07 by XW-HKU