hku-mars/FAST_LIO #14: Long build times for loam_mapping/loam_feature_extract

mbcel, 2021-04-05:
Thank you for that awesome project!

I am experiencing quite long build times for building the project myself within my ros working directory. It takes about 2 minutes at one step. This, of course, would be totally fine if it would just happen ones. However, this happens on any changes that I do to the code with a preceeding build. Like this it is really cumbersome to further code on your project.

Do you also experience this long build times when doing any changes? How do you go about that? Is there any possibility to speed things up? I could not really find out which part is causing this long delay since `catkin_make VERBOSE=1` does not give much more info on that build step.

The actual build step that needs to happen on any code manipulation and that takes rougly 2 minutes: 

`[ 97%] Built target loam_feat_extract `
`// at here it takes about 2 minutes until the [100%] apperas`
`[100%] Linking CXX executable /home/me/catkin_ws/devel/lib/fast_lio/loam_laserMapping`

XW-HKU, 2021-04-23:
Thanks for your interest. But I didn't meet this problem. Would you please tell me the actual cpu type?

mbcel, 2021-05-04:
I use a Nvidia Jetson AGX Xavier which has an ARM-based processor. 

Are you cross-compiling the code or are you directly compiling the code on the target machine (i.e. the dji manifold computer)?

kxch928925819, 2021-05-05:
我直接在dji的mabifold2-c上完成这个程序的所有编译，Ubuntu18.04。其他程序版本，基本与该程序的要求一致！

kxch928925819, 2021-05-05:
你的编译成功了吗？我的manifold2-c已经第二次出现扩展板损坏，原因可能是电路设计的问题，维修了两次。所以，我也在考虑是否更换成你的这个mini计算机，希望能得到你的帮助！
谢谢🙏！

XW-HKU, 2021-05-05:
> I use a Nvidia Jetson AGX Xavier which has an ARM-based processor.
> 
> Are you cross-compiling the code or are you directly compiling the code on the target machine (i.e. the dji manifold computer)?

We are currently working on the implementation on the ARM based platform. The long building time truely exists, several minutes (around 3 minutes) should be taken in my tests.

XW-HKU, 2021-05-05:
> 你的编译成功了吗？我的manifold2-c已经第二次出现扩展板损坏，原因可能是电路设计的问题，维修了两次。所以，我也在考虑是否更换成你的这个mini计算机，希望能得到你的帮助！ 谢谢🙏！
> […](#)

You may consider trying intel NUC, its hardware is more stable than manifold 2c and sells cheaper. Only problem is the weight may be a little bigger than manifold 2c.

mbcel, 2021-05-05:
Hm okay. What puzzels me is that this long build time occures on every code change in the `laserMapping.cpp` file. It seems like the linking process here takes over a minute and needs to be done on ANY change.

Do you experience the same on your ARM platform or do you have such long build times only for the first initial build?

If yes, how do you cope with this? Are you cross-compiling on a different machine or something like that? To wait more than a minute after each code change makes it impossible for me to code properly.

kxch928925819, 2021-05-05:
这个情况我真的没有遇到，我用mabifold2-c进行编译，非常流畅，没有出现这个状况。

kxch928925819, 2021-05-05:
我为接下来准备了mti-300的惯导系统，但是从未标定过imu与lidar，希望能得到你的帮助，标定imu，再将mti-300的数据引入程序，已提高精度！

antun008, 2021-06-04:
what about raspbery pi 4 arm arhitecture?

kxch928925819, 2021-06-04:
没试过这个！

XW-HKU, 2021-07-07:
> what about raspbery pi 4 arm arhitecture?

fast-lio can work on raspberry but remember to close rviz since rviz will consume too much computing resources.