hku-mars/FAST_LIO #16: Does Fast-lio-1.0 support MId-70?

chenxianbo, 2021-05-11:
HI:
Does Fast-lio-1.0 support MId-70?
if it supports, do you have  turning good parameter for Mid-70? parameters is " edgea edgeb smallp_ratio point_filter_num ..."

kxch928925819, 2021-05-11:
这个没试过。
但是horizon我试过，没问题。

------------------ Original ------------------
From: slam_study ***@***.***&gt;
Date: Tue,May 11,2021 8:02 PM
To: hku-mars/FAST_LIO ***@***.***&gt;
Cc: Subscribed ***@***.***&gt;
Subject: Re: [hku-mars/FAST_LIO] Does Fast-lio-1.0 support MId-70? (#16)

chenxianbo, 2021-05-26:
@XW-HKU @kxch928925819 
你好，英语不好，这里用汉字写了。
Hi，my English is pool， so I ask with chinese。

我实现了Fast-lio支持外置的IMU，我有两个问题想咨询下：
I relize Fast-lio support outside IMU， here I want ask two questions：

第一个问题：我想知道代码特征提取函数  void mid_handler(const sensor_msgs::PointCloud2::ConstPtr &msg)  支持Mid-40/70吗，你们有实际测试过吗？如果可以它的参数是多少（是下面的默认参数吗，如果不是能提供一下吗）？

first question：Does function "void mid_handler () " support Mid-40/70? you test with these lidars? If, it support, what is the paramters (Are the default prameters below, or can you give it for me) ?
-------------------------------------------------------------------------------------------------------------------------------------------
n.param("lidar_type", lidar_type, 0);
n.param("blind", blind, 0.5);
n.param("inf_bound", inf_bound, 10);
n.param("N_SCANS", N_SCANS, 1);
n.param("group_size", group_size, 8);
n.param("disA", disA, 0.01);
n.param("disB", disB, 0.1);
n.param("p2l_ratio", p2l_ratio, 400);
n.param("limit_maxmid", limit_maxmid, 9);
n.param("limit_midmin", limit_midmin, 16);
n.param("limit_maxmin", limit_maxmin, 3.24);
n.param("jump_up_limit", jump_up_limit, 175.0);
n.param("jump_down_limit", jump_down_limit, 5.0);
n.param("cos160", cos160, 160.0);
n.param("edgea", edgea, 3);
n.param("edgeb", edgeb, 0.05);
n.param("smallp_intersect", smallp_intersect, 170.0);
n.param("smallp_ratio", smallp_ratio, 1.2);
n.param("point_filter_num", point_filter_num, 4);

--------------------------------------------------------------------------------------------------------------------------------------------
第二个问题：如果使用Mid-40/70，后面的laserMapping的优化参数有需要改的吗？
second question： If use Mid-40/70， is need to change parmeter optimization in laserMapping？

kxch928925819, 2021-05-26:
1、mid-70我没有，但是horizon我测试了，可以支持！
2、mti-300的imu我已经购买了，但是主控的manifold2-c测试中烧了扩展板，下周dji维修才能寄回，没有实现lidar与imu的标定呢！

 

 
 
第一个问题：我想知道代码特征提取函数  void mid_handler(const sensor_msgs::PointCloud2::ConstPtr &amp;msg)  支持Mid-40/70吗，你们有实际测试过吗？如果可以它的参数是多少（是下面的默认参数吗，如果不是能提供一下吗）？

kxch928925819, 2021-05-27:
哈哈！我得找你帮助啦！

关于livox与imu标定 ，能麻烦你给与一定的指导吗！

我这一头雾水！程序都不知道选谁的，调试也不太明白！

我就是一个测绘技术人员，喜欢玩，所以你发的那些调参，我还真的一点忙也帮不上！

这是我的QQ号，希望更多的得到你的帮助！谢谢啊！

XW-HKU, 2021-06-01:
> @XW-HKU @kxch928925819
> 你好，英语不好，这里用汉字写了。
> Hi，my English is pool， so I ask with chinese。
>
> 我实现了Fast-lio支持外置的IMU，我有两个问题想咨询下：
> I relize Fast-lio support outside IMU， here I want ask two questions： [...]

不好意思很晚才回答：
1. FAST LIO1.0 不直接支持MID 70，当然如果有外置IMU，可以修改下imu的topic name看看效果，不过feature extraction不一定能很好的支持mid 70.  FAST LIO 2.0 能比较好的支持外置IMU以及大部分雷达。
2. IMU是必须的。

XW-HKU, 2021-06-01:
> 哈哈！我得找你帮助啦！ 关于livox与imu标定 ，能麻烦你给与一定的指导吗！ 我这一头雾水！程序都不知道选谁的，调试也不太明白！ 我就是一个测绘技术人员，喜欢玩，所以你发的那些调参，我还真的一点忙也帮不上！ 这是我的QQ号，希望更多的得到你的帮助！谢谢啊！
> […](#)

FAST LIO可以外接IMU（修改laser_mapping.cpp里面的ros 订阅topic name），不过需要保证旋转上没有偏差，也就是rotation的外参是单位阵。平移外参可以直接在common_lib.h里面修改。

XW-HKU, 2021-07-07:
Now mid-70 should work with fast-lio if having an external, synced IMU.

closed 2021-07-09 by XW-HKU