hku-mars/FAST_LIO #17: 请问有没有关于仓库里提供的rosbag中imu数据的说明

Saki-Chen, 2021-05-17:
我看了一眼imu数据，静止条件下加速度数据差不多是1，这个数据是有缩放吗？而且也没有看到imu相对于lidar的外参

kxch928925819, 2021-05-17:
目前据我所知，这个程序的imu是内置的博士生产的bmi088 惯导，这个惯导被大多数实验人放弃，我也刚购买了mti-300的惯导，可惜manifold 2-c扩展板烧了，dji更换后，性能下降太厉害，还没解决，实验要等manifold2-c的问题解决后，再进行。

kxch928925819, 2021-05-18:
查看 rostopic 确定一下，imu数据的发布

主题: [hku-mars/FAST_LIO] 请问有没有关于仓库里提供的rosbag中imu数据的说明 (#17)

Saki-Chen, 2021-05-18:
> 目前据我所知，这个程序的imu是内置的博士生产的bmi088 惯导，这个惯导被大多数实验人放弃，我也刚购买了mti-300的惯导，可惜manifold 2-c扩展板烧了，dji更换后，性能下降太厉害，还没解决，实验要等manifold2-c的问题解决后，再进行。
> […](#)

我简单地把imu数据乘上重力加速度，还是能用的

kxch928925819, 2021-05-18:
哈哈，有机会学习一下你的思路！

XW-HKU, 2021-06-01:
> > 目前据我所知，这个程序的imu是内置的博士生产的bmi088 惯导，这个惯导被大多数实验人放弃，我也刚购买了mti-300的惯导，可惜manifold 2-c扩展板烧了，dji更换后，性能下降太厉害，还没解决，实验要等manifold2-c的问题解决后，再进行。
> > […](#)
> 
> 我简单地把imu数据乘上重力加速度，还是能用的

你的做法是正确的，livox系列雷达的内置imu的加速度计数据单位都是重力加速度g。可以参考livox官方wiki：

https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol

（找 IMU data format:）

closed 2021-07-07 by XW-HKU