hku-mars/FAST_LIO #18: 无人机设计图

duanqg, 2021-05-31:
您的工作真的太棒了
我在您的代码的基础上做了一点儿微小的工作，增加了GNSS校准功能，在大场景之下提高了鲁棒性，在进行评估之后会找您进行代码合并。
请问您有这个小型无人机的设计图纸和零件的购买链接么，我也想在进行UAV SLAM验证（ps.horizon特别适合放在无人机上，所以请问第二版什么时候上线哈哈哈

这个是我的一些效果：
[image]

https://user-images.githubusercontent.com/70955391/120224590-52710480-c276-11eb-8149-1eae8372397e.mp4

kxch928925819, 2021-06-01:
哈哈 你很厉害啊 我的horizon也装了这个程序 但是测试结果并没有你的好 原因大概是：我按照geoslam的作业方式进行采集，在路线中有多次停留，将设备放置在地面采集10秒以上的数据，想以此 作为像控点，来提高精度。这样做，估计与程序设定的闭合检查等数据迭代优化方式不符，所以最后闭合差很大！
如果是想在无人机上挂载horizon，可以参考openmms的方案，github上有他们实验小组的代码已经网址！期待你的新结果！同意也和你一样在期待FAST-LIO 2.0！

duanqg, 2021-06-01:
哈哈哈，都是人家的底子打得好
大场景之下确实会出现部分问题
在用GNSS前（两层，没有回环成功）
[image]
用GNSS后（回环成功）
[image]

kxch928925819, 2021-06-01:
我是搞测绘的。  
1、你用的Gps是什么？
2、精度验证了吗？
我试过livox-loam lili-om 还有这个，手里有reach m2 的Gps 还有mti-300的imu ，接下来首先准备完成lidar（livox-horizon）与imu的标定，然后再考虑接入Gps，希望有机会大家一起搞！
谢谢🙏！

XW-HKU, 2021-06-01:
> 您的工作真的太棒了
> 我在您的代码的基础上做了一点儿微小的工作，增加了GNSS校准功能，在大场景之下提高了鲁棒性，在进行评估之后会找您进行代码合并。
> 请问您有这个小型无人机的设计图纸和零件的购买链接么，我也想在进行UAV SLAM验证（ps.horizon特别适合放在无人机上，所以请问第二版什么时候上线哈哈哈
> 
> 这个是我的一些效果：
> [image]
> 
>  2021-06-06.01.11.32.mp4

非常感谢你的关注！期待有一天能把GNSS加入FAST-LIO中。
1. 关于飞机，飞机全部是我自己设计的，源文件应该不会开源，不过我之后会开源一些参考的stl文件，可以进行3d打印。关于这个飞机以及FAST LIO，可以看一下我们组另一个项目 https://github.com/hku-mars/dyn_small_obs_avoidance
2. 第二版其实已经开发完，只是论文还在写，还有一些对照实验需要进行，所以得等文章投出去才能开源。我也想尽快开源，奈何我写文章实在是慢。

XW-HKU, 2021-06-01:
> 我是搞测绘的。   1、你用的Gps是什么？ 2、精度验证了吗？ 我试过livox-loam lili-om 还有这个，手里有reach m2 的Gps 还有mti-300的imu ，接下来首先准备完成lidar（livox-horizon）与imu的标定，然后再考虑接入Gps，希望有机会大家一起搞！ 谢谢🙏！
> […](#)

非常谢谢你，帮我回答了很多问题！我其实个人感觉horizon可能不是很适合测绘，高度方向的精度比较难保证，avia总的来讲可能更好一些。

duanqg, 2021-06-02:
您好，我们已经经过精度评估了。
[image]

如果可以的话，我们实验室有一定的经费，愿意购买您设计的无人机（因为入门实在是太耗精力了）

XW-HKU, 2021-06-03:
> 您好，我们已经经过精度评估了。
> [image]
> 
> 如果可以的话，我们实验室有一定的经费，愿意购买您设计的无人机（因为入门实在是太耗精力了）
> 我的邮箱
> [duanqigeng@gmail.com](mailto:duanqigeng@gmail.com)

之后我们会开源出鸿雁UAV的stl文件，大部分零件可以3D打印制作。不过商业上的事情您可以联系我们导师张富老师 fuzhang@hku.hk。
另外，如果做UAV的话，不太建议使用horizon，z方向的fov太小了，我比较建议用avia。

XW-HKU, 2021-06-03:
> 您的工作真的太棒了
> 我在您的代码的基础上做了一点儿微小的工作，增加了GNSS校准功能，在大场景之下提高了鲁棒性，在进行评估之后会找您进行代码合并。
> 请问您有这个小型无人机的设计图纸和零件的购买链接么，我也想在进行UAV SLAM验证（ps.horizon特别适合放在无人机上，所以请问第二版什么时候上线哈哈哈
> 
> 这个是我的一些效果：
> [image]
> 2021-06-06.01.11.32.mp4

顺便想问下你用的哪一款GNSS？如果方便透露的话，我们用pixhawk的原生gps就感觉数据质量很差。

duanqg, 2021-06-03:
我记得我们使用的是liosam那款小型imu自带的雷达，具体的型号应该是3DM-GX5-45

kxch928925819, 2021-06-03:
3dm-gx5-45
级别不低，我搞了一个gx3-25，咸鱼搞的，价格便宜，但级别太低！现在正在装mti-300，期待好结果！

duanqg, 2021-06-03:
我们也会使用avia进行测绘，现在用的也是的，我们使用GPS基本能达到2cm精度，使用北京唐驿的4Gdtu产品，千寻RTK，在ptp同步的情况下有4hz的GPS输出，比较稳定。我跟姐原来有一定地图导航的LBS服务的经验，现在正在便携针对gps不稳定时的优化筛选算法，但是现在在一些桥梁场景下比较成功，但是在隧道场景就不太行，还在优化评估

duanqg, 2021-06-03:
你可以试试MTI—680G，能便宜一半的价格。也是9轴+gnss，精度差距不大

kxch928925819, 2021-06-03:
哈哈 ，去年就买了reach m2 Gps ，加上mti-300，还有horizon ，我期待能达成测绘级别的slam！ 这周希望能达成livox-imu的标定，试试结果再接入Gps。
我总觉得：如果slam自身的精度不能到一定的程度，过分依赖Gps，就只能服务于汽车导航。

kxch928925819, 2021-06-03:
我咸鱼上收的mti-300，人民币5000元。应该还可以吧！

duanqg, 2021-06-03:
这个不是很清楚

kxch928925819, 2021-06-03:
我用英国的geoslam搞过隧道，一开始怎么都不成功。后来绝望就将各种参数调到最大，50分钟的数据，后处理用了近4个小时。
 所以，如果咱们这个slam的迭代参数，分块大小可以调整，还有就是对后处理时间没有限制，那么就一定可以。
我总是觉得没有完全的必要性，非要达成实时完成，后处理能提高精度的情况下，一定会更有意思。可能这些想法太基于测绘这个小专业了！

duanqg, 2021-06-03:
低速场景更适合双天线rtk，高速更适合单天线rtk。

chengwei0427, 2021-08-06:
你好， @duanqg ，请问你说的`增加了GNSS校准功能，在大场景之下提高了鲁棒性`，是直接增加到FAST-LIO的ESKF内部的，还是增加一个后端优化，在后端优化中使用GNSS校准的？

closed 2023-01-01 by stale[bot]