hku-mars/FAST_LIO #19: Are there any support for ROS2?

Tomato1107, 2021-06-07:
Is there any other way to use raw livox lidar data such as /livox/lidar_xxxxxxxxxx | Type: sensor_msgs/msg/PointCloud2  directly?

XW-HKU, 2021-06-08:
> Is there any other way to use raw livox lidar data such as /livox/lidar_xxxxxxxxxx | Type: sensor_msgs/msg/PointCloud2 directly?

We are trying now.

Tomato1107, 2021-06-08:
Thanks very much!

XW-HKU, 2021-07-07:
I think, if pointcloud2 contains the timestamp of every lidar point (I am not sure about that yet), you may try to use Velodyne configuration to test.

Tomato1107, 2021-09-29:
Thanks for your advice, I will try to check it!

Tomato1107, 2021-11-16:
I have modified the livox_ros_driver to add the offset_time to the PointCloud2 data format, and have tested on fast_lio, thaks very much for your advices!

closed 2021-11-16 by Tomato1107

Tomato1107, 2021-11-28:
> I think, if pointcloud2 contains the timestamp of every lidar point (I am not sure about that yet), you may try to use Velodyne configuration to test.

Here is the performance of the Mid70 with an external IMU without sycn and calibration to the lidar, just adding the offset_time to the pointcloud2.
https://www.youtube.com/watch?v=-t6ghPeJXCE

XW-HKU, 2021-11-28:
> > I think, if pointcloud2 contains the timestamp of every lidar point (I am not sure about that yet), you may try to use Velodyne configuration to test.
> 
> Here is the performance of the Mid70 with an external IMU without sync and calibration to the lidar, just adding the offset_time to the pointcloud2. https://www.youtube.com/watch?v=-t6ghPeJXCE

looks good👍🏻

Tomato1107, 2022-01-11:
> 
Hi! @XW-HKU I have put the modified packages here. To use PointCloud2 format to run the Fast_lio2 directly.
https://github.com/Tomato1107/FAST_LIO_Modified

esaumar, 2023-07-17:
Hi @Tomato1107 !
I'm trying to replicate what you did. I mean, to use the pointcloud2 data format for a livox horizon capture. I have a few questions hoping you can kindly help me:

1. Regarding the modifications of `livox_ros_driver` to add the `offset_time` to the `PointCloud2` data format, did you do something similar to this [repo](https://github.com/juliangaal/livox2pointcloud/blob/28319efc40d10751395ef46747fcedcf42579f43/src/livox2pointcloud.cpp#L19)? 
I mean, did you only add the offset_time field `cloud_msg.fields[0].name = "offset_time";` and then you get the offset_time similar to the function [FillPointsToCustomMsg](https://github.com/Livox-SDK/livox_ros_driver/blob/3d240d5666129e1a3052e78ee8487a04b08fdda3/livox_ros_driver/livox_ros_driver/lddc.cpp#L356) from the original repo?
2. From the FAST-LIO repo, what handler did you use? The avia_handler or the velodyne_handler as suggested by @XW-HKU 
Thank you!
NOTE: Not sure if you have it but it would be really helpful if you already have a forked repo with the modifications you did. :pray: