hku-mars/FAST_LIO #20: Point-to-edge/corner cost in the code and in the paper

lewisjiang, 2021-06-16:
The FAST-LIO described in your paper uses both planar/surf/flat features and edge/corner/sharp features for state estimation. 

In your implementation, the feature processing part extracts and publishes both flat and sharp features [src/feature_extract.cpp#L127#L128](https://github.com/hku-mars/FAST_LIO/blob/6e1fa94351316eafac5fcf55c5fcd575f3e89829/src/feature_extract.cpp#L127). But in state estimation part, it seems that only flat features are subscribed and used [src/laserMapping.cpp#L627](https://github.com/hku-mars/FAST_LIO/blob/6e1fa94351316eafac5fcf55c5fcd575f3e89829/src/laserMapping.cpp#L627). 

May I ask why the sharp features are not used in the implementation?

kxch928925819, 2021-06-17:
如果将面状比作致密基线特征，那么尖锐就只能当作稀疏基线特征，在地下空间、隧道、建筑物等结构中，尖锐特征很容易出现同样的现象，势必造成匹配的混乱！ 
      哈哈！我不是很内行，只能表达一下自己的理解！ 希望能帮到你！

XW-HKU, 2021-07-05:
> 如果将面状比作致密基线特征，那么尖锐就只能当作稀疏基线特征，在地下空间、隧道、建筑物等结构中，尖锐特征很容易出现同样的现象，势必造成匹配的混乱！        哈哈！我不是很内行，只能表达一下自己的理解！ 希望能帮到你！
> […](#)

He is right, the corner feature is not robust in some scenes, we removed the feature extraction and point-to-corner in FAST-LIO2.

lewisjiang, 2021-07-05:
> > 如果将面状比作致密基线特征，那么尖锐就只能当作稀疏基线特征，在地下空间、隧道、建筑物等结构中，尖锐特征很容易出现同样的现象，势必造成匹配的混乱！        哈哈！我不是很内行，只能表达一下自己的理解！ 希望能帮到你！
> > […](#)
> 
> He is right, the corner feature is not robust in some scenes, we removed the feature extraction and point-to-corner in FAST-LIO2.

Thank you for your confirmation. I think you should have mentioned the exclusion of corner features in the paper, or at least mentioned the performance deterioration with corners in experiments as in [this paper](https://ieeexplore.ieee.org/document/8793511), in my opinion.

closed 2021-07-05 by lewisjiang