hku-mars/FAST_LIO #23: 地图的保存

huduo0812, 2021-07-02:
我有这个算法构建了一个非常满意的地方， 可是应该用什么方法保存下来用来定位导航呢？

XW-HKU, 2021-07-05:
We will support saving PCD file soon.

huduo0812, 2021-07-05:
> 我们将很快支持保存 PCD 文件。

好的，谢谢

XW-HKU, 2021-07-06:
现在的算法可以支持保存每一帧点云（世界坐标系下）到pcd文件，可以参考readme去操作。

huduo0812, 2021-07-06:
> 现在的算法可以支持保存每一帧点云（世界坐标系下）到pcd文件，可以参考readme去操作。
非常感谢你们的更新，还有问题需要请教一下，就是，怎么可以像https://github.com/Livox-SDK/livox_mapping这个算法一样一次保存整个地图在一个pcd文件里，然后再转为栅格地图用来定位和导航

XW-HKU, 2021-07-06:
你需要保存哪种形式的地图呢，是所有帧的点云的累积稠密地图？，还是ICP所用到的稀疏点云地图？

kxch928925819, 2021-07-06:
稠密点云，这样抽稀工作可以放在其他软件执行！
再者，有三个个疑问：闭环精度评定的参数能否调整，在程序里怎么调整！
第二，能否直接输出精度评定的数据，图表。
第三，能否通过采集设备静置，得到静置时间段的采集器位置坐标，（譬如程序设定，当imu移动在10秒钟不超过10cm，程序输出该点坐标信息），以便在点云平移旋转时方便处理！
哈哈😄！水平所限，如有冒犯，敬请谅解！ 我就是一个老测量员！

huduo0812, 2021-07-06:
> 你需要保存哪种形式的地图呢，是所有帧的点云的累积稠密地图？，还是ICP所用到的稀疏点云地图？

是的，我想要那种所有帧的累积的地图，然后转成栅格地图，用来室外定位导航。每一帧的点云保存下来都只是局部的吧，我前几天试过修改laserMapping.cpp里的代码，成功保存了稠密的pcd点云地图，但是没有颜色，都是白色，然后想根据intensity渲染一下点云，但是发现intensity都是0，最后根据z轴渲染了一下，非常丑，hhhh

EloyAldao, 2021-07-06:
This is from v1, but worked also on v2.
Add the ap.intensity = pl[j].intensity; and ap.intensity = pl[k].intensity; lines and recompile:

```
 if(j == uint(last_surface+point_filter_num-1))
  {
    PointType ap;
    ap.x = pl[j].x;
    ap.y = pl[j].y;
    ap.z = pl[j].z;
    ap.intensity = pl[j].intensity;
    ap.curvature = pl[j].curvature;
    pl_surf.push_back(ap);

    last_surface = -1;
  }
}
else
{
  if(types[j].ftype==Edge_Jump || types[j].ftype==Edge_Plane)
  {
    pl_corn.push_back(pl[j]);
  }
  if(last_surface != -1)
  {
    PointType ap;
    for(uint k=last_surface; k<j; k++)
    {
      ap.x += pl[k].x;
      ap.y += pl[k].y;
      ap.z += pl[k].z;
      ap.intensity = pl[k].intensity;
      ap.curvature += pl[k].curvature;
    }
    ap.x /= (j-last_surface);
    ap.y /= (j-last_surface);
    ap.z /= (j-last_surface);
    ap.curvature /= (j-last_surface);
    pl_surf.push_back(ap);
  }
  last_surface = -1;
```

XW-HKU, 2021-07-06:
> > 你需要保存哪种形式的地图呢，是所有帧的点云的累积稠密地图？，还是ICP所用到的稀疏点云地图？
> 
> 是的，我想要那种所有帧的累积的地图，然后转成栅格地图，用来室外定位导航。每一帧的点云保存下来都只是局部的吧，我前几天试过修改laserMapping.cpp里的代码，成功保存了稠密的pcd点云地图，但是没有颜色，都是白色，然后想根据intensity渲染一下点云，但是发现intensity都是0，最后根据z轴渲染了一下，非常丑，hhhh

最新的代码已经改成了保存成一个累积的点云，现在rivz里面的当前帧点云上色使用的就是intensity，你可以看看保存出来的pcd file有没有intensity.

XW-HKU, 2021-07-06:
> This is from v1, but worked also on v2.
> Add the ap.intensity = pl[j].intensity; and ap.intensity = pl[k].intensity; lines and recompile:
>
> ```
>  if(j == uint(last_surface+point_filter_num-1))
>   { [...]

Good Jobs, I will add these into codes. Thank you very much!

XW-HKU, 2021-07-06:
> 稠密点云，这样抽稀工作可以放在其他软件执行！ 再者，有三个个疑问：闭环精度评定的参数能否调整，在程序里怎么调整！ 第二，能否直接输出精度评定的数据，图表。 第三，能否通过采集设备静置，得到静置时间段的采集器位置坐标，（譬如程序设定，当imu移动在10秒钟不超过10cm，程序输出该点坐标信息），以便在点云平移旋转时方便处理！ 哈哈smile！水平所限，如有冒犯，敬请谅解！ 我就是一个老测量员！
> […](#)

实在不好意思，测绘的这些我不太懂，感觉应该可以在点云后处理里面实现？

XW-HKU, 2021-07-06:
> Good Jobs, I will add these into codes. Thank you very much!

Already merged in the codes.

huduo0812, 2021-07-06:
> 最新的代码已经改成了保存成一个累积的点云，现在rivz里面的当前帧点云上色使用的就是intensity，你可以看看保存出来的pcd file有没有intensity.

好的，谢谢你们，我明天试试

huduo0812, 2021-07-07:
> 最新的代码已经改成了保存成一个累积的点云，现在rivz里面的当前帧点云上色使用的就是intensity，你可以看看保存出来的pcd file有没有intensity.

[image]
已经保存成 累积的点云的，但是颜色和rviz里面看到的不太一样，

XW-HKU, 2021-07-07:
> [image]
> 已经保存成 累积的点云的，但是颜色和rviz里面看到的不太一样，

change what to visualize/color by pressing 1,2,3,4,5.
1 is all random
2 is X values
3 is Y values
4 is Z values
5 is intensity

huduo0812, 2021-07-07:
> change what to visualize/color by pressing 1,2,3,4,5.
> 1 is all random
> 2 is X values
> 3 is Y values
> 4 is Z values
> 5 is intensity

谢谢谢，现在效果非常好
[image]

closed 2021-07-07 by XW-HKU

GRF-Sunomikp31, 2023-07-30:
@huduo0812 Hi，想咨询下你，如果利用保存的PCD点云文件重新定位呢？