hku-mars/FAST_LIO #24: FastLio-2: catkin_make & ikd-Tree Failed (因ikd-Tree编码失败？)

Noahcuptea, 2021-07-05:
### 尊敬的港大团队好：

先恭喜恭喜，新作品的进展让人激动。

这次尝试编译Fast Lio2遇到困难， 请多多指教 😀：

#### 0. 安装

在说明文稿 “2.build” 里的`git clone https://github.com/XW-HKU/fast_lio.git`, 
可能要改为：`git clone https://github.com/hku-mars/FAST_LIO.git`

**1. 第一次编译**,不论是用终端clone，还是直接下载都没有**Ikd-treed**文件夹内的文件，导致编译失败。可能是文件夹名存在@符号造成的？

`简化的报错信息如下`：
``` CMD
make[2]: *** 没有规则可制作目标“
~  /ikd-Tree/ikd_Tree.cpp”，

fatal error: ikd-Tree/ikd_Tree.h: 
没有那个文件或目录 #include <ikd-Tree/ikd_Tree.h>
```

**2. 再次尝试编译**, 把Ikd-treed的档案手动copy到文件夹，CMD输入catkin_make有大量的报错。怀疑是编码问题 或 cmake版本问题，多多指教啊~ 

报错档案如下， 共三类。

- 第一类是编码报错

`简化的报错信息如下`：
```CMD
[ 29%] Building CXX object FAST_LIO/ ... /ikd_Tree.cpp.o

~ ikd-Tree/ikd_Tree.cpp:70:76: 

error: stray ‘\302’ in program
   <title>ikd-Tree/ikd_Tree.cpp at 3d115a41377243420a74fc15dd7cf7ef337730df 
�� hku-mars/ikd-Tree · GitHub</title>
```

- 第二类是语法报错

`简化的报错信息如下`：
```CMD
~ ikd_Tree.cpp:320:12: 

warning: missing terminating ' character
       <!-- '"` --><!-- </textarea></xmp> --></option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="341097199" data-scoped-search-url="/hku-mars/ikd-Tree/search" data-owner-scoped-search-url="/orgs/hku-mars/search" data-unscoped-search-url="/search" action="/hku-mars/ikd-Tree/search" accept-charset="UTF-8" method="get">
```
- 第三类似乎是建议改写

`简化的报错信息如下`：
```CMD

~ /laserMapping.cpp:859:82: 
error: ‘kdtree_incremental_time’ was not declared in this scope
                 
aver_time_incre = 
aver_time_incre * (frame_num - 1)/frame_num + (kdtree_incremental_time)/frame_num;

------------------------------------------
~/laserMapping.cpp:859:82: 

note: suggested alternative: ‘map_incremental’
                
 aver_time_incre = 
aver_time_incre * (frame_num - 1)/frame_num + (kdtree_incremental_time)/frame_num;
                                                                                 
```

lewisjiang, 2021-07-05:
You should `cd FAST_LIO/` before `git submodule update --init`.

XW-HKU, 2021-07-05:
> You should `cd FAST_LIO/` before `git submodule update --init`.

二楼说的对，需要先cd进FAST_LIO再git submodule update --init。

XW-HKU, 2021-07-05:
不好意思，说明文档还在撰写中，近期会逐步update。

Noahcuptea, 2021-07-05:
> You should `cd FAST_LIO/` before `git submodule update --init`.

收到，安装顺利完成，谢谢！

Noahcuptea, 2021-07-05:
> 不好意思，说明文档还在撰写中，近期会逐步update。

已经很好了！ 

我这边是直接下载zip解压后，用git submodule update --init没有效果。
一定是git clone 才可以用git submodule

已安装好。太谢谢了！

closed 2021-07-05 by Noahcuptea

XW-HKU, 2021-07-05:
我修改了readme，修正了你提到的错误。还有其他问题请立即通知我。

duongthao1218, 2022-03-08:
```
$ ~cd FAST_LIO/include/
$ git clone https://github.com/hku-mars/ikd-Tree.git
```