hku-mars/FAST_LIO #25: MID-40/70 with IMU

Noahcuptea, 2021-07-05:
尊敬的港大团队好：

因为avia型号买不到，只能尝试用mid-40和70绑定外接imu，遇到些困难，请指点下~

- 尝试把loam_livox里，mid系列的launch_file 拿到fast_lio尝试。可以看到点云，但优化可能不明显。
您会考虑 **mid**系列外接imu 的优化么？

- 还试了按过去的[相关回复](https://github.com/hku-mars/FAST_LIO/issues/16)修改，但mid系列的点云没有出现在rviz界面里。请问还适用么？
```
相关回复：`Does Fast-lio-1.0 support MId-70?` 
FAST LIO可以外接IMU（修改laser_mapping.cpp里面的ros 订阅topic name），
不过需要保证旋转上没有偏差，也就是rotation的外参是单位阵。
平移外参可以直接在common_lib.h里面修改。
```

XW-HKU, 2021-07-06:
改一下avia.yaml的lid topic和imu topic，然后直接跑fastlio2的mapping_avai.launch，会怎样

Noahcuptea, 2021-07-06:
> 改一下avia.yaml的lid topic和imu topic，然后直接跑fastlio2的mapping_avia.launch，会怎样

尝试修改avia.yaml之后，rviz没反应(版用是是您刚上传的pcd saving新版本)。

推测问题出在IMU数据格式，过程如下：

**0. 排除连接问题**：
分别测试了imu和lidar，功能正常。
也确定FastLio成功订阅了imu和lidar的topics。

```CMD
livox@livox-vpc:~$ rostopic info /handsfree/imu
Type: sensor_msgs/Imu

Publishers: 
 * /imu (http://livox-vpc:39337/)

Subscribers: 
 * /laserMapping (http://livox-vpc:38929/)
```
```CMD
livox@livox-vpc:~$ rostopic info /livox/lidar
Type: livox_ros_driver/CustomMsg

Publishers: 
 * /livox_lidar_publisher (http://livox-vpc:34407/)

Subscribers: 
 * /laserMapping (http://livox-vpc:38929/)
```

**1. avia.yaml里lidar的订阅部分**

AVIA和MID系列topic都是 **/livox/lidar**

FastLio2和Loam_livox的区别是选用ros_driver里不同的数据格式:

- Fastlio2 - 览沃自定义点云数据(`livox_lidar_msg.launch`)

- Loam_Livox -  pointcloud2 格式的点云数据 (`livox_lidar.launch`)

都是官方数据格式，感觉没问题。

**2.avia.yaml里imu的订阅部分**

更改avia.yaml的imu topic后，猜测可能是Avia imu和外接imu的数据格式不同，导致rviz没反应。

如果是下列imu数据格式的问题，请指点下怎么修改哈😀~

- 外接imu数据格式：
``` CMD
livox@livox-vpc:~$ rosmsg show sensor_msgs/Imu
std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
geometry_msgs/Quaternion orientation
  float64 x
  float64 y
  float64 z
  float64 w
float64[9] orientation_covariance
geometry_msgs/Vector3 angular_velocity
  float64 x
  float64 y
  float64 z
float64[9] angular_velocity_covariance
geometry_msgs/Vector3 linear_acceleration
  float64 x
  float64 y
  float64 z
float64[9] linear_acceleration_covariance
```

- AVIA内置bmi088 imu数据格式：
<html>
<body>
<p style="box-sizing: border-box; margin-bottom: 16px; margin-top: 0px; color: rgb(36, 41, 46); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"> </p>

Field | Offset (byte) | Data Type | Description
-- | -- | -- | --
gyro_x | 0 | float | Unit: rad/s
gyro_y | 4 | float | Unit: rad/s
gyro_z | 8 | float | Unit: rad/s
acc_x | 12 | float | Unit: g
acc_y | 16 | float | Unit: g
acc_z | 20 | float | Unit: g

<br class="Apple-interchange-newline">
</body>
</html>

XW-HKU, 2021-07-06:
1. 目前，fastlio只支持使用 livox_lidar_msg.launch采集的livox数据，因为只有livox_lidar_msg.launch提供每个雷达点的时间戳，livox_lidar.launch没有每个点的时间戳，这个对点云去畸变非常重要。
2. IMU的话只要是 sensor_msgs/Imu 类型就行。

Noahcuptea, 2021-07-06:
> 1. 目前，fastlio只支持使用 livox_lidar_msg.launch采集的livox数据，因为只有livox_lidar_msg.launch提供每个雷达点的时间戳，livox_lidar.launch没有每个点的时间戳，这个对点云去畸变非常重要。
> 2. IMU的话只要是 sensor_msgs/Imu 类型就行。

又试了几次，RViz没有cloud point，也没有imu的移动。

您这边考虑做mid系列外挂imu的测试么?

XW-HKU, 2021-07-06:
有个很重要的问题，你的IMU和雷达之间做了时间同步吗？

XW-HKU, 2021-07-06:
你可以考虑发一个rosbag sample给我

Noahcuptea, 2021-07-06:
> 有个很重要的问题，你的IMU和雷达之间做了时间同步吗？

您建议用哪种方式做时间同步，请给指条路线吧😀~

我了解到的有PTP时间同步法：
- [《Livox Time Sync 教程》](https://github.com/Livox-SDK/Livox-SDK/wiki/livox-device-time-synchronization-manual-cn)好像能做lidar和系统时间的同步，这种方式也能做lidar和imu的同步么？ 
--- 详细教程 [《OUTSTER的PTP时间同步教程》，看84页](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2p0.pdf)

除了PTP的方法，您还推荐别的方法么？怕研究半天，走了死路 哈哈。

XW-HKU, 2021-07-06:
你先看下lidar topic的时间戳和imu的时间戳差多少。
同步的话，具体怎么做我也不是很清楚。

Noahcuptea, 2021-07-07:
> 你先看下lidar topic的时间戳和imu的时间戳差多少。
> 同步的话，具体怎么做我也不是很清楚。

坚果云链接里是rosbag包, 您过目:
https://www.jianguoyun.com/p/DQYeZcQQsP6nCBiuk_8D 

PPT时间同步要求网卡硬件支持IEEE 1588, 其它的同步类型包括pps和gnss都有硬件要求.

且不能保证主机时间和lidar同步后, imu时间也能和lidar时间同步, 暂时没收集到案例.

理论上知道主机时间, lidar和imu的时间戳应该可以做时间同步, 好像没有未知量, 咱们用什么路线去做呢?

XW-HKU, 2021-07-07:
> > 你先看下lidar topic的时间戳和imu的时间戳差多少。
> > 同步的话，具体怎么做我也不是很清楚。
> 
> 坚果云链接里是rosbag包, 您过目:
> https://www.jianguoyun.com/p/DQYeZcQQsP6nCBiuk_8D
> 
> PPT时间同步要求网卡硬件支持IEEE 1588, 其它的同步类型包括pps和gnss都有硬件要求.
> 
> 且不能保证主机时间和lidar同步后, imu时间也能和lidar时间同步, 暂时没收集到案例.
> 
> 理论上知道主机时间, lidar和imu的时间戳应该可以做时间同步, 好像没有未知量, 咱们用什么路线去做呢?

硬件同步必须要有硬件支持，一般的外接IMU都自带一个上位机发送同步信号，如果没有的话就只能自己搭建一个上位机了。

软件同步的话，最简单的是你另外写个程序，把lidar 时间和IMU时间都修正到同一时间基准上。但是软件同步无法处理通信延迟。因为，通信时间是不稳定的而且对于imu和lidar是不一样的。

Noahcuptea, 2021-07-07:
> 
> 硬件同步必须要有硬件支持，一般的外接IMU都自带一个上位机发送同步信号，如果没有的话就只能自己搭建一个上位机了。
> 
> 软件同步的话，最简单的是你另外写个程序，把lidar 时间和IMU时间都修正到同一时间基准上再publish出来。但是软件同步精度有限，无法处理通信延迟。因为通信时间是不稳定的，而且对于imu和lidar是不一样的。

谢谢您的耐心~因为要买设备, 继续测试算法, 再跟您确认下: 

咱们rosbag里, Rviz没有point cloud和imu位姿的原因,
是因为imu和lidar没时间同步, 是这样么?

如果确定问题出在同步, 我会按照下图做时间同步, uINS国内好像也没货, 所以要买别家的惯导. 

您给看看图后面惯导的购买标准对不对. 

[image]

惯导购买标准:

- 支持rs485协议, 或ttl(ttl可以转rs485)
- 提供时间戳 (时间数据有要求么?)
- imu的发送格式必须是单位阵? (这里不太确定)
- 还有什么要注意的么?

XW-HKU, 2021-07-07:
> 谢谢您的耐心~因为要买设备, 继续测试算法, 再跟您确认下:
>
> 咱们rosbag里, Rviz没有point cloud和imu位姿的原因,
> 是因为imu和lidar没时间同步, 是这样么?
>
> 如果确定问题出在同步, 我会按照下图做时间同步, uINS国内好像也没货, 所以要买别家的惯导.
>
> 您给看看图后面惯导的购买标准对不对.
>
> [image]
>
> 惯导购买标准:
>
>     * [...]

Your data can work, just pull the newest code and follow the updated readme.

imu data format is just the standard sensor_msgs/Imu.

For hardware sync, I don't know too much details.

Noahcuptea, 2021-07-08:
> Your data can work, just pull the newest code and follow the updated readme.
>
> imu data format is just the standard sensor_msgs/Imu.
>
> For hardware sync, I don't know too much details.

十分感谢! 我做好硬同步测试后, 再给您反馈~

XW-HKU, 2021-07-08:
you can set time_sync_en to true to test your data，before hardware sync is ready

Noahcuptea, 2021-07-08:
> 你现在就可以试试打开time_sync_en，看看mid70的效果如何。

不好意思回复晚了, 在找硬同步方案和供应商~ 

Rosbag在下面链接里: 
https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D 

avia.yaml 配置如下:
[image]

测试机MID-70 & MID-40, Rviz暂无点云, Terminal提示如下: 

```
IMU and LiDAR not Synced, 
IMU time: 1625739751.841357, 
lidar scan end time: 0.000000

IMU and LiDAR not Synced, 
IMU time: 1625739751.943336, 
lidar scan end time: 219.498690

IMU and LiDAR not Synced, 
IMU time: 1625739752.043455, 
lidar scan end time: 219.598530

..... 
```

XW-HKU, 2021-07-08:
我这儿可以直接跑链接里面的数据，你是不是代码没有更新？

[image: mid70.png]

Noahcuptea, 2021-07-08:
> 我这儿可以直接跑链接里面的数据，你是不是代码没有更新？

能跑， 我写错个东西， 稍等我做个完整测试~

Noahcuptea, 2021-07-08:
> 你现在就可以试试打开time_sync_en，看看mid70的效果如何。

**Rosbag和录像在下面链接**：
https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D

**测试机体是mid-70** 
没有做imu和lidar的位置的准确标定，imu在lidar左半边上方如下图：
[image]

**代码能跑，墙体似乎比loam版本更平直，点云更清晰，还发现如下点**：

- x和y轴较稳定

- 扫描出的点云体有时轻微错位重叠，怀疑是lidar&imu sync没有做硬同步的原因

- z轴变动超过15度左右，会产生大范围漂移 (看视频第40s： ROSBAG和video.mp4 )

- 实际操作和z轴数据上下相反：往上抬高机器，rviz里雷达会往下扫；往下压机器，rviz里雷达会向上扫 (同样看第40s视频： ROSBAG和video.mp4 )。猜测可能是avia.yaml里，单位矩阵或Z做标符号填写的原因，如下图：
[image]

XW-HKU, 2021-07-08:
> **Rosbag和录像在下面链接**：
> https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D
>
> **测试机体是mid-70**
> 没有做imu和lidar的位置的准确标定，imu在lidar左半边上方如下图：
> [image]
>
> **代码能跑，墙体似乎比loam版本更平直，点云更清晰，还发现如下点**：
>
>     * x和y轴较稳定
>
>     * [...]

看上去，你的mid 70的xy轴和IMU的xy轴没有对齐，很有可能IMU的xy轴和mid70的是反的。

Noahcuptea, 2021-07-09:
> xy轴对齐了吗？

对, 上个测试xy没对齐.

**下链接是是对齐后的Rosbag和pcd**:
https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D 

**MID70&imu测试: 稳定和准确度能达到良好**: 

- 注意开始采集要对着大空间, 如果是不足2m的简单空间, 定位会漂移

- 虽然rosbag play会有相同点云区域的错位重合, 但是pcd是良好的, 采集成功

- 一个几分钟的测量, 基本的墙体位置信息都有了  
[image]

**下一步是lidar&imu的外参标定**
如果调整好lidar&imu的外参标定, 效果一定更好. 

您指导下, 我忙完会回来继续做测试. 

咱们translational extrinsic这个变量是外参标定么? 
变量extrinsic_T: [ 0.04165, 0.02326, -0.0284 ] 是xyz, 单位是mm么? 
原点大概在lidar哪个位置?

XW-HKU, 2021-07-09:
extrinsic_T 是平移外参，以IMU作为原点，雷达相对于IMU的位置偏差，雷达自身的坐标参考mid70的手册。

对于室内小场景，可以试试把launch里面的 filter_size_surf 和 filter_size_map 调低到0.3或者0.2

vipelz, 2021-07-14:
I am a young architect, i just bought a Mid70 and I'm using It after a lot of effort with this https://github.com/Livox-SDK/livox_scanner 
it would be perfect for someone like me, to use MID 70 + IMU to obtain point clouds for indoor environments.

Could you please share in English how to combine Mid 70 + IMU ?
Also which IMU product are you using ?

  Thanks so much guys !!

XW-HKU, 2021-07-15:
> I am a young architect, i just bought a Mid70 and I'm using It after a lot of effort with this https://github.com/Livox-SDK/livox_scanner
> it would be perfect for someone like me, to use MID 70 + IMU [...]

Thanks for your interests. Have you ever followed the section 3.2 in the readme of Fast-lio?

vipelz, 2021-07-15:
> Thanks for your interests. Have you ever followed the section 3.2 in the readme of Fast-lio?
Yes Sir but I don't have the IMU yet so I'll buy one.
So I'd like to know which model you tested to be sure it will fit without any problems.
  Thank you

XW-HKU, 2021-07-16:
> > Thanks for your interests. Have you ever followed the section 3.2 in the readme of Fast-lio?
> 
> Yes Sir but I don't have the IMU yet and I'm going to buy one.
> So I'd like to know which model you tested to be sure it will fit without any problems.
> Thank you

I should be sorry to say that I didn't build a actual mid-70 with external IMU system. I tested the data from the external IMU with Velodyne LiDAR and Livox horizon, such as the data from LIO-SAM and LILI-OM. Theoratically the FAST-LIO can support MID-70 with external IMU if the IMU can 1. support the time-sync and 2. produce the rostopic with the type sensor_msgs/Imu. I think the second requirement is easy to satisfy. About the first requirement, I think you can refer to the IMU used in the LILI-OM dataset: **Xsens MTi-670**. If it can work with Livox horizon, I think it can work with mid serials, but I cannot guarantee since I am not an expert on this part.

vipelz, 2021-07-18:
The MTi-670 is a little expensive for me.  
I would be happy if it also works with other slightly cheaper IMUs.  
Otherwise I'll have to prepare the piggy bank.

Thank you!!

Noahcuptea, 2021-09-01:
> extrinsic_T 是平移外参，以IMU作为原点，雷达相对于IMU的位置偏差，雷达自身的坐标参考mid70的手册。
> 
> 对于室内小场景，可以试试把launch里面的 filter_size_surf 和 filter_size_map 调低到0.3或者0.2

**您好，这一个半月在做硬同步和各种尺度的测试，如下**：

### 软同步

#### 宿舍校区立面设计 [中小设计]
`10,000 sqm，不算沉降，跟施工图比较，高度相差300mm，误差在5%左右。`

https://user-images.githubusercontent.com/45952143/131610125-431af316-78a6-441e-bece-cf7c58ffa8a0.mp4

#### 居家室内设计 [小设计]
`200 sqm, 软同步在小尺度下不稳，各种漂移`

影片略。。。

### 硬同步

#### 街区立面设计 [中设计]
`300m长步行街，记录了复杂的街区外貌`

https://user-images.githubusercontent.com/45952143/131614892-33ff5dca-e021-427f-9636-2e1a3d597a30.mp4

#### 美容院室内设计 [小设计]
`100± sqm, 3分钟测量完，精准大幅提高`

从室外透过窗户量室内，因窗户折射可能有重影
[image]

手工测量图
[image]

雷达出图
[image]
.
.
**(7980-7975)/7980 = 0.0006 = 0.06%** 

#### 楼梯封盖 [小微设计]
`2.5± sqm, 在点云上建模，出图给供货商`

在点云上建模
[image]

依靠模型出图
[image]
[image]

### 后续
请问fastlio能给点云赋色么？ 如果能通过镜头给每个点赋色，就可看出材质，能让更多人使用Fastlio. 
您给指导下怎么给点云赋色的技术路线，谢谢~

XW-HKU, 2021-09-01:
> > extrinsic_T 是平移外参，以IMU作为原点，雷达相对于IMU的位置偏差，雷达自身的坐标参考mid70的手册。
> > 对于室内小场景，可以试试把launch里面的 filter_size_surf 和 filter_size_map 调低到0.3或者0.2
> 
> **您好，这一个半月在做硬同步和各种尺度的测试，如下**：
> ### 软同步
> #### 宿舍校区立面设计 [中小设计]
> 
> `10,000 sqm，不算沉降，跟施工图比较，高度相差300mm，误差在5%左右。`
> goodsuniversity.x264.aac.mp4
> #### 居家室内设计 [小设计]
> 
> `200 sqm, 软同步在小尺度下不稳，各种漂移`
> 
> 影片略。。。
> ### 硬同步
> #### 街区立面设计 [中设计]
> 
> `300m长步行街，记录了复杂的街区外貌`
> Rec.2021-09-01.13.02.31.mp4
> #### 美容院室内设计 [小设计]
> 
> `100± sqm, 3分钟测量完，精准大幅提高`
> 
> 从室外透过窗户量室内，可能因窗户折射出现了重影
> [image]
> 
> 手工测量图
> [image]
> 
> 雷达出图
> [image]
> .
> .
> **(7980-7975)/7980 = 0.0006 = 0.06%**
> #### 楼梯封盖 [小微设计]
> 
> `2.5± sqm, 在点云上建模，出图给供货商`
> 
> 在点云上建模
> [image]
> 
> 依靠模型出的两版图
> [image]
> [image]
> 
> 现场图：
> [[image]](url)
> ### 后续
> 
> 请问fastlio能给点云赋色么？ 如果能通过镜头给每个点赋色，就可看出材质，能让更多人使用Fastlio.
> 您给指导下怎么给点云赋色的技术路线，谢谢~

谢谢分享！
现在的点云里面有intensity的值，可以根据intensity着色。由于激光（都是单一波长/频率）原理的限制，除了intensity 外，没有其他颜色信息可以提供。

Noahcuptea, 2021-09-01:
> 谢谢分享！
> 现在的点云里面有intensity的值，可以根据intensity着色。由于激光（都是单一波长/频率）原理的限制，除了intensity 外，没有其他颜色信息可以提供。

有没有可能跟camera做外参，借用camera取色，把RGB颜色附在点云 xyz坐标点 后面？

XW-HKU, 2021-09-01:
> > 谢谢分享！
> > 现在的点云里面有intensity的值，可以根据intensity着色。由于激光（都是单一波长/频率）原理的限制，除了intensity 外，没有其他颜色信息可以提供。
> 
> 有没有可能跟camera做外参，借用camera取色，把RGB颜色附在点云 xyz坐标点后面？

如果要融合相机，可以参考我们组的R2LIVE。

vipelz, 2021-09-01:
Could you please share 

> > extrinsic_T is the external translation parameter, with the IMU as the origin, the position deviation of the radar relative to the IMU, and the coordinates of the radar itself refer to the manual of mid70.
> > For small indoor scenes, try reducing the filter_size_surf and filter_size_map in launch to 0.3 or 0.2
> 
> **Hello, I have been doing hard synchronization and various scale tests for this month and a half, as follows** :
> 
> ### Soft sync
> #### Facade design of dormitory campus [small and medium design]
> `10,000 sqm，不算沉降，跟施工图比较，高度相差300mm，误差在5%左右。`
> 
>  goodsuniversity.x264.aac.mp4 
> #### Home Interior Design [Small Design]
> `200 sqm, 软同步在小尺度下不稳，各种漂移`
> 
> The video is omitted. . .
> 
> ### Hard sync
> #### Block Elevation Design [Chinese Design]
> `300m长步行街，记录了复杂的街区外貌`
> 
>  Rec.2021-09-01.13.02.31.mp4 
> #### Beauty salon interior design [small design]
> `100± sqm, 3分钟测量完，精准大幅提高`
> 
> Measure indoors from the outside through the windows, there may be ghosting due to window refraction
> [image]
> 
> Manual measurement chart
> [image]
> 
> Radar showing . . **(7980-7975) / 7980 = 0.0006 = 0.06%**
> [image]
> 
> 
> ****
> 
> #### Stair Cover [Small Micro Design]
> `2.5± sqm, 在点云上建模，出图给供货商`
> 
> Modeling on a point cloud
> [image]
> 
> Two layouts based on the model
> [image]
> [image]
> 
> Site map:
> [[image]](url)
> 
> ### Follow-up
> Can fastlio color the point cloud? If you can color each point through the lens, you can see the material, allowing more people to use Fastlio.
> You can give the technical route of how to color the point cloud under the guidance, thank you~

Could you please share your set-up?
For example what is your IMU model, what about  settings and how you build It? 
Video and photo are better than 100 words!
Thank you!!

vipelz, 2021-09-03:
你能分享一下你的设置吗？
 例如，您的 IMU 模型是什么，设置如何以及您如何构建它？
 视频和照片胜过100字！
 谢谢！！

Could you please share your set-up?
For example what is your IMU model, what about settings and how you build It?
Video and photo are better than 100 words!
Thank you!!

luameows, 2022-01-10:
@Noahcuptea 您好，方便分享下lidar与imu之间外参标定的方法么？我尝试了lidar_align的方法，对于纯imu计算的里程计会有较大偏差导致标定结果不是很好，您有其他的方法建议么？

closed 2023-01-01 by stale[bot]

5204338, 2023-07-25:
> * 虽然rosbag play会有相同点云区域的错位重合, 但是pcd是良好的, 采集成功
> *

大佬你好，我想请教以下，雷达和IMU的时间同步是怎么做的啊

wyb17js, 2023-12-28:
> > * 虽然rosbag play会有相同点云区域的错位重合, 但是pcd是良好的, 采集成功
> 
> 大佬你好，我想请教以下，雷达和IMU的时间同步是怎么做的啊

你好，请问你学会时间同步的方法了吗？我使用的是livox mid360雷达，但是我并不知道该如何将它进行时间对齐