hku-mars/FAST_LIO #26: [ INFO] [1625547852.411847770]: IMU Initializing: 5.0 %

kxch928925819, 2021-07-06:
~~~/home/dji/catkin_ws/src/FAST_LIO/ file opened
[ INFO] [1625547852.411847770]: IMU Initializing: 5.0 %
[ WARN] [1625547852.415338028]: Reset ImuProcess
[ INFO] [1625547852.415483487]: IMU Initials: Gravity: 0.5451 0.0623 -9.7936 0.9901; state.bias_g: 0.0001 0.0001 0.0001; acc covarience: 0.10000000 0.10000000 0.10000000; gry covarience: 0.10000000 0.10000000 0.10000000
FAST-LIO not ready

IMU Initializing: 5.0 %?
FAST-LIO not ready

XW-HKU, 2021-07-06:
which launch file? is IMU and lidar synchronized?

kxch928925819, 2021-07-06:
roslaunch fast_lio mapping_avia.launch

kxch928925819, 2021-07-06:
dji@dji-MANIFOLD-2-C:~/catkin_ws$ roslaunch fast_lio mapping_avia.launch
... logging to /home/dji/.ros/log/5e1c8074-de1a-11eb-8842-34d26269e60d/roslaunch-dji-MANIFOLD-2-C-9130.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dji-MANIFOLD-2-C:45925/

SUMMARY
========

PARAMETERS
 * /common/imu_topic: /livox/imu
 * /common/lid_topic: /livox/lidar
 * /cube_side_length: 1000.0
 * /dense_map_enable: True
 * /feature_extract_enable: False
 * /filter_size_map: 0.5
 * /filter_size_surf: 0.5
 * /mapping/acc_cov: 0.1
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 450.0
 * /mapping/extrinsic_R: [1, 0, 0, 0, 1, 0...
 * /mapping/extrinsic_T: [0.04165, 0.02326...
 * /mapping/fov_degree: 90
 * /mapping/gyr_cov: 0.1
 * /max_iteration: 3
 * /point_filter_num: 3
 * /preprocess/blind: 1
 * /preprocess/lidar_type: 1
 * /preprocess/scan_line: 6
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /runtime_pos_log_enable: False

NODES
  /
    laserMapping (fast_lio/fastlio_mapping)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [9140]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5e1c8074-de1a-11eb-8842-34d26269e60d
process[rosout-1]: started with pid [9151]
started core service [/rosout]
process[laserMapping-2]: started with pid [9158]
process[rviz-3]: started with pid [9159]
Multi thread started 
p_pre->lidar_type 1
~~~~/home/dji/catkin_ws/src/FAST_LIO/ file opened
[ INFO] [1625549048.718784550]: IMU Initializing: 5.0 %
[ WARN] [1625549048.721875512]: Reset ImuProcess
[ INFO] [1625549048.722061005]: IMU Initials: Gravity: 0.0789 0.5868 -9.7911 0.9914; state.bias_g: 0.0001 0.0001 0.0001; acc covarience: 0.10000000 0.10000000 0.10000000; gry covarience: 0.10000000 0.10000000 0.10000000
FAST-LIO not ready
^C[rviz-3] killing on exit
[laserMapping-2] killing on exit
[ WARN] [1625549063.634023184]: catch sig 2
Rebuild thread terminated normally
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
dji@dji-MANIFOLD-2-C:~/catkin_ws$

XW-HKU, 2021-07-06:
which bag are you test? or in real avia device?

XW-HKU, 2021-07-06:
can you see the rviz and point cloud?

kxch928925819, 2021-07-06:
Yes! Rviz is Ok!

XW-HKU, 2021-07-06:
ok, if point cloud looks fine, then ignore the output "FAST-LIO not ready", I will fix that right now

kxch928925819, 2021-07-06:
HAHA~! Tankyou very much! In  the afternoon ,I start this over with the desktop,Tommrow ,underground parking test it !

XW-HKU, 2021-07-06:
It's fixed, please git pull than test again

closed 2021-07-06 by XW-HKU