hku-mars/FAST_LIO #27: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]

JasperTan97, 2021-07-06:
I get the following error. Does anyone know the fix?

>  error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
>          omp_set_num_threads(MP_PROC_NUM);
>

XW-HKU, 2021-07-06:
1. Varify if the openmp is installed, if not, apt install it:

 apt-get install libomp-dev

2. try catkin_make, not catkin build

JasperTan97, 2021-07-06:
Hi @XW-HKU Thank you for your reply. I have installed openmp and done catkin_make. However I still get the error:

```
home/isera2/catkin_ws/src/FAST_LIO/src/laserMapping.cpp: In function ‘void h_share_model(state_ikfom&, esekfom::dyn_share_datastruct<double>&)’:
/home/isera2/catkin_ws/src/FAST_LIO/src/laserMapping.cpp:559:40: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
         omp_set_num_threads(MP_PROC_NUM);
                                        ^
In file included from /home/isera2/catkin_ws/src/FAST_LIO/src/laserMapping.cpp:35:0:
/usr/lib/gcc/x86_64-linux-gnu/7/include/omp.h:86:13: note:   initializing argument 1 of ‘void omp_set_num_threads(int)’
 extern void omp_set_num_threads (int) __GOMP_NOTHROW;

```

XW-HKU, 2021-07-06:
What's your ubuntu version?

JasperTan97, 2021-07-06:
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0 

I use ROS melodic, Eigen version 3.3.4. 

I can run Fast_lio 1 but I cannot catkin_make this new version.

XW-HKU, 2021-07-06:
A bit confusing, do you run git submodule update --init?

XW-HKU, 2021-07-06:
may be you can try delete the devel/ and build/ directory before catkin_make

JasperTan97, 2021-07-06:
> A bit confusing, do you run git submodule update --init?
Yes I did. I have the ikd-tree files. 

> may be you can try delete the devel/ and build/ directory before catkin_make
I have cleaned and remade the folder many times.

I can't find where MP_PROC_NUM is defined in laserMapping.cpp In omp.h, the function takes in an int datatype, which I think is the source of the error. How was MP_PROC_NUM defined?

Once again, I like to thank you for your prompt responses!

XW-HKU, 2021-07-06:
> Yes I did. I have the ikd-tree files.
>
>
> I have cleaned and remade the folder many times.
>
> I can't find where MP_PROC_NUM is defined in laserMapping.cpp In omp.h, the function takes in an int [...]

The MP_PROC_NUM is defined in CMakeLists.txt. It is predefined during cmake based on your cpu archtectuer.

Oh, what's your cpu type?

JasperTan97, 2021-07-06:
cat /proc/cpuinfo  | grep 'name'| uniq
model name	: Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz
 cat /proc/cpuinfo  | grep process| wc -l
4

XW-HKU, 2021-07-06:
When you fresh making the project, you will see the messages in terminator like below:

> -- The C compiler identification is GNU 7.5.0
> -- The CXX compiler identification is GNU 7.5.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Using CATKIN_DEVEL_PREFIX: /home/xw/XW/ws_fastlio_hku/devel
> -- Using CMAKE_PREFIX_PATH: /home/xw/XW/ws_fastlio_hku/devel;/home/xw/ws_common/devel;/home/xw/ws_livoxrosdriver/devel;/opt/ros/melodic
> -- This workspace overlays: /home/xw/XW/ws_fastlio_hku/devel;/home/xw/ws_common/devel;/home/xw/ws_livoxrosdriver/devel;/opt/ros/melodic
> -- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
> -- Using PYTHON_EXECUTABLE: /usr/bin/python2
> -- Using Debian Python package layout
> -- Using empy: /usr/bin/empy
> -- Using CATKIN_ENABLE_TESTING: ON
> -- Call enable_testing()
> -- Using CATKIN_TEST_RESULTS_DIR: /home/xw/XW/ws_fastlio_hku/build/test_results
> -- Found gtest sources under '/usr/src/googletest': gtests will be built
> -- Found gmock sources under '/usr/src/googletest': gmock will be built
> -- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
> -- Looking for pthread.h
> -- Looking for pthread.h - found
> -- Looking for pthread_create
> -- Looking for pthread_create - not found
> -- Looking for pthread_create in pthreads
> -- Looking for pthread_create in pthreads - not found
> -- Looking for pthread_create in pthread
> -- Looking for pthread_create in pthread - found
> -- Found Threads: TRUE  
> -- Using Python nosetests: /usr/bin/nosetests-2.7
> -- catkin 0.7.29
> -- BUILD_SHARED_LIBS is on
> -- BUILD_SHARED_LIBS is on
> -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> -- ~~  traversing 1 packages in topological order:
> -- ~~  - fast_lio
> -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> -- +++ processing catkin package: 'fast_lio'
> -- ==> add_subdirectory(FAST_LIO)
> Current CPU archtecture: x86_64
> Processer number:  16
> core for MP:  3

You can show me the last three lines to see if there is any difference.

> Current CPU archtecture: x86_64
> Processer number:  16
> core for MP:  3

JasperTan97, 2021-07-06:
Current CPU archtecture: x86_64
Processer number:  4
core for MP:  2

XW-HKU, 2021-07-06:
> Current CPU archtecture: x86_64
> Processer number: 4
> core for MP: 2

That should be fine. what's your cmake version?

JasperTan97, 2021-07-06:
cmake version 3.10.2

XW-HKU, 2021-07-06:
> cmake version 3.10.2

Seems everything is right.

JasperTan97, 2021-07-06:
Hi, I made it work.

I replaced 
`    add_definitions(-DMP_PROC_NUM="${PROC_NUM}")`
in the CMakeLists.txt with 
`    add_definitions(-DMP_PROC_NUM=2)`
directly and it works fine. I'm not really sure what's the issue? Maybe the syntax is wrong or there is a bug with the typecasting but it now works. 

Thanks so much for your time. We can close the issue if there's nothing else to add.

XW-HKU, 2021-07-06:
> MP_PROC_NUM

Thank you, there is truly a bug in cmake file, I fixed it in the newly update.

closed 2021-07-06 by XW-HKU