hku-mars/FAST_LIO #28: How compute H ???

duandongbin000, 2021-07-06:
Dear friend:
   我直接说中文啦！我是个slam初学者，看了您们的论文“FAST-LIO: A Fast, Robust LiDAR-inertial Odometry Package by Tightly-Coupled Iterated Kalman Filter”，有个地方很迷惑,，不知道论文公式14中雅可比H怎么计算的？具体什么形式呢？
[image]

XW-HKU, 2021-07-06:
根据error state的定义：真值 x_k = propagated x_k \boxplus  error-state x_k，h_j 函数中的真值 x_k可以替换成 ( propagated x_k \boxplus  error-state x_k), 由于error-state x_k很小，所以可以根据泰勒公式在将原的h_j（真值 x_k, {L_j} n_{f_j}）在 h_j（propagated x_k, {L_j} n_{f_j}）处进行线性化（等于说“在error-state x_k = 0处进行线性化”）。

因此 H矩阵的物理含义就是h_j函数对error-state x_k的Jacobin matrix。

duandongbin000, 2021-07-08:
这个H是对称三角阵吗？能否给个H的具体什么形式 。  H = P_cross * R * G + G ???

XW-HKU, 2021-07-09:
> 这个H是对称三角阵吗？能否给个H的具体什么形式 。 H = P_cross * R * G + G ???

首先你要知道hj()是一个一维的数。

H_j是H矩阵的第j行，H_j 是一个1×18的向量但是只有前6个不为0，前三个是对旋转的偏导，第四到第六个是对位移的偏导。具体的解析形式我有空了会写一个。

duandongbin000, 2021-07-09:
谢啦 非常感谢您的指导。

XW-HKU, 2021-07-13:
Sorry fot the late reply.

For delta p, I think it is right.

But for delta R, looks like a bit weired. delta R should be a R^3 vector not a rotation matrix. You can follow the below guide:

[image]

duandongbin000, 2021-07-13:
okay， thank you！

xiaotaw, 2021-09-29:
@XW-HKU 我推出来的公式和代码中的不太一样，请帮忙看看。
[image]

cuishiwei, 2022-05-14:
> @XW-HKU 我推出来的公式和代码中的不太一样，请帮忙看看。 [image]

你好，请教一下这个求导过程，最后怎么理解的？你照片中确实和代码有出入啊。

xiaotaw, 2022-05-15:
@cuishiwei 代码和论文是一致的，形式上差一个转置。我之前手写的可能推错了，没有校验。后面有一个电子版的，供参考。

[image]

简略的推导过程

[image]

yangtzey, 2022-08-08:
@xiaotaw 你好，请问下方便把详细推导文档分享下吗？谢谢！ 841581437@qq.com

xiaotaw, 2022-08-08:
> @xiaotaw 你好，请问下方便把详细推导文档分享下吗？谢谢！ [841581437@qq.com](mailto:841581437@qq.com)

之前截图这版有误，修正之后的版本，整理了一下，放到了知乎上，排版稍微有点差，暂时没有时间去优化，勉强也能看。https://zhuanlan.zhihu.com/p/538975422

yangtzey, 2022-08-09:
非常感谢！

------------------

江hill水
***@***.***

Send time: Monday, Aug 8, 2022 10:34 PM