hku-mars/FAST_LIO #30: 里程计信息，

huduo0812, 2021-07-08:
启动了mapping_avia.launch会发布里程计信息，我现在正在给机器狗做定位导航，机器狗没有编码器，所以需要利用雷达和imu得出里程计信息，请问有什么方法可以只发布里程计信息吗

XW-HKU, 2021-07-08:
the rostopic   /aft_mapped_to_init   is the odometry output

huduo0812, 2021-07-08:
> the rostopic /aft_mapped_to_init is the odometry output

有只发布里程计信息的功能包吗，mapping_avia.launch这个运行起来太占资源

XW-HKU, 2021-07-08:
太占资源是因为开着rviz显示点云吧，你可以关掉rviz里面的 点云显示 或者直接关掉rviz。点云发布并不占资源，显示点云更加吃资源。

huduo0812, 2021-07-08:
> 太占资源是因为开着rviz显示点云吧，你可以关掉rviz里面的 点云显示 或者直接关掉rviz。点云发布并不占资源，显示点云非常占资源。

好的，我试试

XW-HKU, 2021-07-08:
> > 太占资源是因为开着rviz显示点云吧，你可以关掉rviz里面的 点云显示 或者直接关掉rviz。点云发布并不占资源，显示点云非常占资源。
> 
> 好的，我试试

试完请告诉我有没有变化。

huduo0812, 2021-07-08:
> > > 太占资源是因为开着rviz显示点云吧，你可以关掉rviz里面的 点云显示 或者直接关掉rviz。点云发布并不占资源，显示点云非常占资源。
> > 
> > 
> > 好的，我试试
> 
> 试完请告诉我有没有变化。

我把rviz那个参数设置成false，运行起来好一些，但是资源占用还是比较多的，而且还在一直发布点云和地图话题，这是不需要的话题
[image]
现在只运行了livox_lidar_msg.launch  和  mapping_avia.launch  你们可以出一个功能包把雷达和imu发布里程计信息这个功能独立出来，很多项目是没有编码器的，无法提供导航包的里程计信息，非常感谢你的帮助！！

XW-HKU, 2021-07-08:
> > > > 太占资源是因为开着rviz显示点云吧，你可以关掉rviz里面的 点云显示 或者直接关掉rviz。点云发布并不占资源，显示点云非常占资源。
> > > 
> > > 
> > > 好的，我试试
> > 
> > 
> > 试完请告诉我有没有变化。
> 
> 我把rviz那个参数设置成false，运行起来好一些，但是资源占用还是比较多的，而且还在一直发布点云和地图话题，这是不需要的话题
> [image]
> 现在只运行了livox_lidar_msg.launch 和 mapping_avia.launch 你们可以出一个功能包把雷达和imu发布里程计信息这个功能独立出来，很多项目是没有编码器的，无法提供导航包的里程计信息，非常感谢你的帮助！！

谢谢你的建议，我已经在launch file里面加了一个scan_publish_enable的开关，设置为0就会关掉点云publish。

但是有点奇怪，FASTLIO2理论上跑在ARM上的时候最多只会占用3个线程。你这个截图里面6个线程都这么高占用率着实不太对劲。

huduo0812, 2021-07-08:
> 谢谢你的建议，我已经在launch file里面加了一个scan_publish_enable的开关，设置为0就会关掉点云publish。
>
> 但是有点奇怪，FASTLIO2理论上跑在ARM上的时候最多只会占用3个线程。你这个截图里面6个线程都这么高占用率着实不太对劲。

可能是livox_lidar_msg.launch也占用了，我现在下载一下你们新的代码试试

XW-HKU, 2021-07-09:
> 可能是livox_lidar_msg.launch也占用了，我现在下载一下你们新的代码试试

ARM上面CPU占用率过高的问题现在还有吗？

huduo0812, 2021-07-09:
> ARM上面CPU占用率过高的问题现在还有吗？

[image]

我把rviz和dense_publish_enable和scan_publish_enable都设置成0了，这样就只发布了里程计信息，cpu占用率没那么高了

XW-HKU, 2021-07-09:
> [image]
>
> 我把rviz和dense_publish_enable和scan_publish_enable都设置成0了，这样就只发布了里程计信息，cpu占用率没那么高了

可否帮忙测试下，没有livox_lidar_msg.launch，也就是只播放bag的时候，cpu占用情况是怎样的？

huduo0812, 2021-07-11:
[image]
这是我什么都没运行的时候 cpu占用率从十几到二十几一直跳，
[image]
[image]

这是运行bag的时候，占有率是30-60跳

XW-HKU, 2021-07-11:
> [image]
> 这是我什么都没运行的时候 cpu占用率从十几到二十几一直跳，
> [image]
> [image]
> 
> 这是运行bag的时候，占有率是30-60跳

ok

closed 2021-07-11 by XW-HKU