hku-mars/FAST_LIO #31: Adapt for use with a specific external IMU

Jaetriel, 2021-07-08:
Hi, so I would like to try this package, I've been having decent results with Ouster's built-in IMU but I wanna try it using a Microstrain IMU that I have. The problem is the sensor has different coordinates for acceleration and attitude so I'm not sure how to get all the transformations needed correctly in one matrix. Is there a way to adapt the code to have 2 separate ext matrices? Mine would look something like this:
extrinsicRot: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1]
  extrinsicRPY: [0, 1, 0,
                 -1, 0, 0,
                  0, 0, 1]
There should be a way to make this possible but I'm not exactly sure where to start as I am not yet familiar with FAST-LIO

XW-HKU, 2021-07-08:
Well,i think you only need to consider the acceleration and gyroscope, these two should have totally the same coordinates. Fastlio does not use any attitude quaternion from imu, So no need to consider attitude extrinsic.

But should be care that the next parameter in fastlio is defined as lidar wrt imu. May be different with other LIO.

Jaetriel, 2021-07-08:
@XW-HKU thanks for the quick answer, but I am not sure what you mean by `lidar wrt imu`. Does that mean that you transform from lidar frame to IMU frame and not the other way around?

XW-HKU, 2021-07-08:
> @XW-HKU thanks for the quick answer, but I am not sure what you mean by `lidar wrt imu`. Does that mean that you transform from lidar frame to IMU frame and not the other way around?

The lidar's pose under the imu's body frame.

Jaetriel, 2021-07-08:
I changed the extrinsicR matrix to 
extrinsic_R: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1] and that seems to have fixed the movement and acceleration now all work well, but my robot orientation is still not pointed in the right direction

XW-HKU, 2021-07-08:
The global frame of FAST-LIO is defined as the first IMU's frame. That means the IMU's x-y-z axis at the very beginning will be the global frame's x-y-z axis. So the robot's orientation may not looks like your imagine.

closed 2021-07-09 by Jaetriel