hku-mars/FAST_LIO #32: Horizon

kxch928925819, 2021-07-09:
<launch>

	<arg name="rviz" default="true" />

	<rosparam command="load" file="$(find fast_lio)/config/horizon.yaml" />

	<param name="feature_extract_enable" type="bool" value="0"/>
	<param name="point_filter_num" type="int" value="3"/>
	<param name="max_iteration" type="int" value="3" />
	<param name="scan_publish_enable" type="bool" value="0" />
	<param name="dense_publish_enable" type="bool" value="1" />
	<param name="filter_size_surf" type="double" value="0.5" />
	<param name="filter_size_map" type="double" value="0.5" />
	<param name="cube_side_length" type="double" value="1000" />
	<param name="runtime_pos_log_enable" type="bool" value="1" />
	<param name="pcd_save_enable" type="bool" value="1" />
    <node pkg="fast_lio" type="fastlio_mapping" name="laserMapping" output="screen"  /> 

	<group if="$(arg rviz)">
	<node launch-prefix="nice" pkg="rviz" type="rviz" name="rviz" args="-d $(find fast_lio)/rviz_cfg/loam_livox.rviz" />
	</group>

</launch>

<param name="pcd_save_enable" type="bool" value="1" />  but pcd file has not issues!

XW-HKU, 2021-07-09:
did you update your code?

kxch928925819, 2021-07-09:
yes！Brfore catkin-make，edit launch！

kxch928925819, 2021-07-09:
[ WARN] [1625819309.222937980]: TF_OLD_DATA ignoring data from the past for frame aft_mapped at time 1.6258e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1625819309.239976639]: TF_OLD_DATA ignoring data from the past for frame aft_mapped at time 1.6258e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ mapping ]: time: IMU + Map + Input Downsample: 0.004825 ave match: 0.006561 ave solve: 0.001344  ave ICP: 0.012392  map incre: 0.000452 ave total: 0.012847 icp: 0.007939 construct H: 0.000437 
[ mapping ]: time: IMU + Map + Input Downsample: 0.003644 ave match: 0.006562 ave solve: 0.001344  ave ICP: 0.011889  map incre: 0.000490 ave total: 0.012847 icp: 0.007940 construct H: 0.000437 
[ WARN] [1625819309.424248839]: TF_OLD_DATA ignoring data from the past for frame aft_mapped at time 1.6258e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ mapping ]: time: IMU + Map + Input Downsample: 0.003654 ave match: 0.006562 ave solve: 0.001344  ave ICP: 0.012661  map incre: 0.000419 ave total: 0.012848 icp: 0.007941 construct H: 0.000437

XW-HKU, 2021-07-09:
maybe you can set runtime_pos_log_enable to 0, that one may lead to problem. 

If problem still exists, use avia.launch instead, these two has little difference.

kxch928925819, 2021-07-09:
[ WARN] [1625819309.424248839]: TF_OLD_DATA ignoring data from the past for frame aft_mapped at time 1.6258e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained

XW-HKU, 2021-07-09:
`<rosparam command="load" file="$(find fast_lio)/config/horizon.yaml" />

	<param name="feature_extract_enable" type="bool" value="0"/>
	<param name="point_filter_num" type="int" value="3"/>
	<param name="max_iteration" type="int" value="3" />
	<param name="scan_publish_enable" type="bool" value="1" />
	<param name="dense_publish_enable" type="bool" value="1" />
	<param name="filter_size_surf" type="double" value="0.5" />
	<param name="filter_size_map" type="double" value="0.5" />
	<param name="cube_side_length" type="double" value="1000" />
	<param name="runtime_pos_log_enable" type="bool" value="0" />
	<param name="pcd_save_enable" type="bool" value="1" />
    <node pkg="fast_lio" type="fastlio_mapping" name="laserMapping" output="screen"  /> 
`

I test this, PCD got.

kxch928925819, 2021-07-09:
链接：https://pan.baidu.com/s/1_gllUZ7-BG5eYF-AnzsyAA 
提取码：evzr

kxch928925819, 2021-07-09:
小徐：拜托抽空看一下，其他人如果感兴趣，也希望能一起交流。
我们的设备是 livox-horizon  以及manifold 2-c ，电池采用sony摄像机的V口电池，支架采用的斯坦尼康的支架，还有一个背心暂时未登场，这样支架直接受力在肩背，可以轻而易举的取下，进行像控点的采集。

XW-HKU, 2021-07-09:
好，一会儿有空了我看下

kxch928925819, 2021-07-09:
谢谢！

XW-HKU, 2021-07-09:
你发的数据包太大了，我这边下载速度很慢。

你说的scan_publish关掉后没有pcd的问题，我在最新代码里面fix了。现在在关掉点云publish的事后，也会有pcd save。但是如果点云太大，可能pcd save会出问题。我没试过很大的pcd保存，你可以试试。

主要是fastlio更多的是用于定位（给机器人无人机使用），你说的静止坐标提取这个不太好加入现有代码，会和其他模块冲突。

GPS的问题，我们正在添加，但是还需要一段时间。

kxch928925819, 2021-07-09:
”你说的静止坐标提取这个不太好加入现有代码“ 关于这句话，是否可以理解为如果在采集过程中，静置10秒，会对结果造成比较大的影响？

XW-HKU, 2021-07-09:
> ”你说的静止坐标提取这个不太好加入现有代码“ 关于这句话，是否可以理解为如果在采集过程中，静置10秒，会对结果造成比较大的影响？

我个人觉得，这个功能最好是在拿到轨迹数据之后的后处理里来做，放在轨迹生成的地方比较难实现。不过你们自己可以修改代码试试。

kxch928925819, 2021-07-09:
非常感谢！把注意力放在轨迹数据和后处理！太谢谢啦！

XW-HKU, 2021-07-09:
> 非常感谢！把注意力放在轨迹数据和后处理！太谢谢啦！

客气

kxch928925819, 2021-07-12:
/**************** save map ****************/
    /* 1. make sure you have enough memories
    /* 2. pcd save will largely influence the real-time performences **/
    if (pcl_wait_save->size() > 0 && pcd_save_en)
    {
        string file_name = string("scans.pcd");
        string all_points_dir(string(string(ROOT_DIR) + "PCD/") + file_name);
        pcl::PCDWriter pcd_writer;
        cout << "current scan saved to /PCD/" << file_name<<endl;
        pcd_writer.writeBinary(all_points_dir, *pcl_wait_save);
    }
what save scans(time or  cout).pcd ,please!

kxch928925819, 2021-07-12:
dji@dji-MANIFOLD-2-C:~/catkin_ws/src/FAST_LIO/Log$ python3 plot.py
Traceback (most recent call last):
  File "plot.py", line 3, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'
dji@dji-MANIFOLD-2-C:~/catkin_ws/src/FAST_LIO/Log$ 

imu.txt is empty!
  wo hope you  issues  (evo_traj euroc utm kitti ) which is ?

XW-HKU, 2021-07-12:
> dji@dji-MANIFOLD-2-C:~/catkin_ws/src/FAST_LIO/Log$ python3 plot.py Traceback (most recent call last): File "plot.py", line 3, in import numpy as np ModuleNotFoundError: No module named 'numpy' dji@dji-MANIFOLD-2-C:~/catkin_ws/src/FAST_LIO/Log$
> 
> imu.txt is empty!
> wo hope you issues (evo_traj euroc utm kitti ) which is ?

you may consider comment the line 36-51 in the plot.py.

kxch928925819, 2021-07-19:
我总是觉得，咱们的rviz显示，把坐标系统设置的不太舒服，譬如：我想要按照视场角进行采集时，显示的点云界面，很难进行旋转，设置，而如果是反向，就是把imu重力方向倒过来，会发现比较容易实现！
不知道说的对不对，麻烦你看看！

XW-HKU, 2021-07-19:
> 我总是觉得，咱们的rviz显示，把坐标系统设置的不太舒服，譬如：我想要按照视场角进行采集时，显示的点云界面，很难进行旋转，设置，而如果是反向，就是把imu重力方向倒过来，会发现比较容易实现！
> 不知道说的对不对，麻烦你看看！

你可以设置一下rviz，Panel->view->invert z

kxch928925819, 2021-07-19:
/**************** save map ****************/
    /* 1. make sure you have enough memories
    /* 2. pcd save will largely influence the real-time performences **/
    if (pcl_wait_save->size() > 0 && pcd_save_en)
    {
        string file_name = string("scans.pcd");
        string all_points_dir(string(string(ROOT_DIR) + "PCD/") + file_name);
        

string file_name = string("scans.pcd") 
replace 
        static size_t i=0;
        string file_name = string("scans_")+to_string(i)+string(".pcd");

save laserMapping.cpp ,next to catkin_make , scans.pcd will scans01.pcd .it can count!

kxch928925819, 2021-07-20:
lasermapping.cpp  

// This is an advanced implementation of the algorithm described in the
// following paper:
//   J. Zhang and S. Singh. LOAM: Lidar Odometry and Mapping in Real-time.
//     Robotics: Science and Systems Conference (RSS). Berkeley, CA, July 2014.

// Modifier: Livox               dev@livoxtech.com

// Copyright 2013, Ji Zhang, Carnegie Mellon University
// Further contributions copyright (c) 2016, Southwest Research Institute
// All rights reserved.
//
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions are met:
//
// 1. Redistributions of source code must retain the above copyright notice,
//    this list of conditions and the following disclaimer.
// 2. Redistributions in binary form must reproduce the above copyright notice,
//    this list of conditions and the following disclaimer in the documentation
//    and/or other materials provided with the distribution.
// 3. Neither the name of the copyright holder nor the names of its
//    contributors may be used to endorse or promote products derived from this
//    software without specific prior written permission.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
// AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
// IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
// ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
// LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
// CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
// SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
// INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
// CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
// ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
// POSSIBILITY OF SUCH DAMAGE.
#include <omp.h>
#include <mutex>
#include <math.h>
#include <thread>
#include <fstream>
#include <csignal>
#include <unistd.h>
#include <Python.h>
#include <so3_math.h>
#include <ros/ros.h>
#include <Eigen/Core>
#include "IMU_Processing.hpp"
#include <nav_msgs/Odometry.h>
#include <nav_msgs/Path.h>
#include <visualization_msgs/Marker.h>
#include <pcl_conversions/pcl_conversions.h>
#include <pcl/point_cloud.h>
#include <pcl/point_types.h>
#include <pcl/filters/voxel_grid.h>
#include <pcl/io/pcd_io.h>
#include <sensor_msgs/PointCloud2.h>
#include <tf/transform_datatypes.h>
#include <tf/transform_broadcaster.h>
#include <geometry_msgs/Vector3.h>
#include <livox_ros_driver/CustomMsg.h>
#include "preprocess.h"
#include <ikd-Tree/ikd_Tree.h>
#include <stdio.h>
#include <time.h>
#include <string>
#include<fstream>

#define INIT_TIME           (0.1)
#define LASER_POINT_COV     (0.001)
[... 913 lines cut ...]
    }

    fout_out.close();
    fout_pre.close();

    if (runtime_pos_log)
    {
        vector<double> t, s_vec, s_vec2, s_vec3, s_vec4, s_vec5, s_vec6, s_vec7;    
        FILE *fp2;
        string log_dir = root_dir + "/Log/fast_lio_time_log.csv";
        fp2 = fopen(log_dir.c_str(),"w");
        fprintf(fp2,"time_stamp, total time, scan point size, incremental time, search time, delete size, delete time, tree size st, tree size end, add point size, preprocess time\n");
        for (int i = 0;i<time_log_counter; i++){
            fprintf(fp2,"%0.8f,%0.8f,%d,%0.8f,%0.8f,%d,%0.8f,%d,%d,%d,%0.8f\n",T1[i],s_plot[i],int(s_plot2[i]),s_plot3[i],s_plot4[i],int(s_plot5[i]),s_plot6[i],int(s_plot7[i]),int(s_plot8[i]), int(s_plot10[i]), s_plot11[i]);
            t.push_back(T1[i]);
            s_vec.push_back(s_plot9[i]);
            s_vec2.push_back(s_plot3[i] + s_plot6[i]);
            s_vec3.push_back(s_plot4[i]);
            s_vec5.push_back(s_plot[i]);
        }
        fclose(fp2);
    }

    return 0;
}

XW-HKU, 2021-07-21:
非常感谢！我之后看下如何合并进来。新疆好地方呀！

closed 2023-01-01 by stale[bot]