hku-mars/FAST_LIO #33: Postprocess optimization

EloyAldao, 2021-07-09:
Dear Cris:

I use to have a low cpu consuming yaml file for data capture (recording a bag file) and a highly optimized for accuracy yaml file for postprocessing the bag recorded in field.
My question, regarding this, is:

- **point_filter_num**: I understand that this downsamples the pointcloud, grouping the number of points indicated in the parameter. Is it right? If I want the best result should I use value 1?
- **max_iteration**: this referes to the maximum iterations on ICP adjustment per scan. Is it right? So I should use around 10?
- **filter_size_surf**: This I think that is the voxelization size. So 0.5 means that every plane of 0.5x0.5 should be used in the adjustment. Is it right?
- **filter_size_map**: This value I have no idea what means.

If I want the best fitting, even playing the bag at rate 0.5 or 0.1:

- What parameters do you recomend for the best accuracy? 
- Should I modify any of the ones hardcoded in the sources?

Thanks in advance

XW-HKU, 2021-07-09:
point_filter_num: You can set to 1 of course, the run time will be double i think.
    max_iteration: the influence is very tiny, I actually dont recommend to set it to large.
    filter_size_surf: I always use same one as filter_size_map but you can try to set this to a different and smaller value.
    filter_size_map: This value it the voxel size in the map. for indoor small environments, you may consider to set to 0.2-0.3, but in outdoor large scale scenes, smaller than 0.4 may lead to worse performences.

I think now there are no more hard code parameters you need modify for better accuracy.

EloyAldao, 2021-07-09:
Ok.
Thank you very much!!

closed 2021-07-09 by EloyAldao