hku-mars/FAST_LIO #34: save trajectory from scan

Jaetriel, 2021-07-09:
Hi, is there a way currently to save the trajectory from a scan in something like a pcd file like the map is saved?

XW-HKU, 2021-07-09:
you can try to set runtime_pos_log_enable to 1，then you can find a txt here：/Log/pos_log.txt.

ly-uuu, 2022-05-25:
> you can try to set runtime_pos_log_enable to 1，then you can find a txt here：/Log/pos_log.txt.
> 
> For the file format, please refer to dump_lio_state_to_log(FILE *fp) in src/LaserMapping.cpp.

您好，请问怎么输出类似于TUM/KITTI的数据格式呢

closed 2023-01-01 by stale[bot]

wangchuanhao1231, 2023-10-31:
> > 你可以尝试将 runtime_pos_log_enable 设置为 1，然后你可以在这里找到一个 txt：/Log/pos_log.txt。
> > 文件格式请参考 src/LaserMapping.cpp 中的 dump_lio_state_to_log（FILE *fp）。
> 
> 您好，请问怎么输出类似于TUM/KITTI的数据格式呢

你好，请问你知道如何输出了吗