hku-mars/FAST_LIO #36: Question about discrete model in fast lio

LeisureLei, 2021-07-12:
Thanks for your great job! I test fast lio2 on KITTI dataset and get nice results. 
But I have a question about discrete model when I read fast lio paper. 
[image]
According to formula (2) (3), we can get P_{t+1} = P_{t} + V_{t} * δt , why not P_{t+1} = P_{t} + V_{t} * δt + 1/2 * a_{t} * δt^{2}?

XW-HKU, 2021-07-12:
Thanks for your interests. You are right. But fastlio1 uses the zero order discretion，fastlio2 uses the first order model. But they are really very similar since the imu sample time is always very tiny.

LeisureLei, 2021-07-13:
> Thanks for your interests. You are right. Fastlio1 uses the zero order discretion，Fastlio2 uses the first order model. But they are really very similar since the imu sample time is always very tiny.

Thanks! And I have another question about the computation of mattix F_x and F_w in appendix A. Are there some materials describing specific process?

XW-HKU, 2021-07-13:
> Thanks! And I have another question about the computation of mattix F_x and F_w in appendix A. Are there some materials describing specific process?

I think the computation of the F_x and F_w would be very easy, if you regard appendix A as a simple chain-rule of partial differential. And [this paper](https://arxiv.org/pdf/2102.03804.pdf)'s appendix has a simple example for the computation of chain-rule of partial differential combining the boxplus operation, may help you.

LeisureLei, 2021-07-13:
> > > Thanks for your interests. You are right. Fastlio1 uses the zero order discretion，Fastlio2 uses the first order model. But they are really very similar since the imu sample time is always very tiny.
> > 
> > 
> > Thanks! And I have another question about the computation of mattix F_x and F_w in appendix A. Are there some materials describing specific process?
> 
> I think the computation of the F_x and F_w would be very easy, if you regard appendix A as a simple chain-rule of partial differential. And [IKFOM](https://arxiv.org/pdf/2102.03804.pdf)'s appendix A&B has a simple example for the computation of chain-rule of partial differential combining the boxplus operation, may help you.

Thanks!

closed 2021-07-13 by LeisureLei

LeisureLei, 2021-07-15:
@XW-HKU hi, I wonder what the difference  about matrix F_x and F_w between **fast lio** and **Quaternion kinematics for the error-state KF**. I try to compare them. And some parts sre same, some parts are different.

XW-HKU, 2021-07-15:
> @XW-HKU hi, I wonder what the difference about matrix F_x and F_w between **fast lio** and **Quaternion kinematics for the error-state KF**. I try to compare them. And some parts sre same, some parts are different.

The main differences i think is using rotation matrix or quaternion.

LeisureLei, 2021-07-15:
> > @XW-HKU hi, I wonder what the difference about matrix F_x and F_w between **fast lio** and **Quaternion kinematics for the error-state KF**. I try to compare them. And some parts sre same, some parts are different.
> 
> The main differences i think is using rotation matrix or quaternion.

Got it!

HaisenbergPeng, 2022-04-05:
> > @XW-HKU hi, I wonder what the difference about matrix F_x and F_w between **fast lio** and **Quaternion kinematics for the error-state KF**. I try to compare them. And some parts sre same, some parts are different.
> 
> The main differences i think is using rotation matrix or quaternion.

Hi, Mr. Xu, there is a very good [blog](https://zhuanlan.zhihu.com/p/441182819) about ESKF deduction based on SO3, which is written by Dr. Gao, Xiang.  Based on his deduction, there should be no "-A(w_i*deltaT)^T*deltaT" for equation 7  in your 2021 RAL fastlio paper.
[image]
What do you think?

ZongHengScholar, 2023-05-07:
> 
I'm also curious that it's not in the code, but it's in the paper.Again, i should be clear about the formula.

ZongHengScholar, 2023-05-07:
> Hi, Mr. Xu, there is a very good [blog](https://zhuanlan.zhihu.com/p/441182819) about ESKF deduction based on SO3, which is written by Dr. Gao, Xiang. Based on his deduction, there should be no [...]

I'm also curious that it's not in the code, but it's in the paper.Again, i should be clear about the formula.

zhouqiuhua, 2024-09-29:
> Thanks for your great job! I test fast lio2 on KITTI dataset and get nice results. But I have a question about discrete model when I read fast lio paper. [image] According to formula (2) (3), we can get P_{t+1} = P_{t} + V_{t} * δt , why not P_{t+1} = P_{t} + V_{t} * δt + 1/2 * a_{t} * δt^{2}?
Hello boss, may I ask which sequences from the Kitti dataset you tested on Fast-LiO2? I tested all the sequences, but only 05, 06, 07, and 09 were successful. The others were all quite outrageous

zhouqiuhua, 2024-09-29:
Hello boss, may I ask which sequences from the Kitti dataset you tested on Fast-LiO2? I tested all the sequences, but only 05, 06, 07, and 09 were successful. The others were all quite outrageous

Orangeflying, 2025-06-10:
> 谢谢你的出色工作！我在 KITTI 数据集上测试了 Fast lio2，得到了不错的结果。 但是我在阅读 Fast lio 论文时，对离散模型有一个疑问。 根据公式 (2) (3)，我们可以得到 P_{t+1} = P_{t} + V_{t} * δt ，为什么不是 P_{t+1} = P_{t} + V_{t} * δt + 1/2 * a_{t} * δt^{2} 呢？ [image]

您好，请问您在kitti数据集上测试了哪些序列呢？我在测试过程中发现fastlio2出现较大的定位漂移