hku-mars/FAST_LIO #37: Integration with GNSS

salovision, 2021-07-12:
First of all, thanks for the great work & research you've done on SLAM, this is definitely the best implementation so far what comes to mapping precision and performance!

I have a GNSS receiver which outputs centimeter accurate position information at 8 Hz synced with lidar & imu clock, and I'd like to try to improve the long-run odometry drift by fusing GNSS location.

What would be the best approach to do that?  Is it possible to enter external position estimate from GNSS to Kalman filter?

Optimally, I'd want to fuse GNSS location when it's available (outdoors, with proper GNSS fix) up to 8 Hz.  But when it's indoors or with poor GNSS reception, I'd use the existing prediction system.

Also, what would be the best way to alter the initial rotation (heading/yaw) so that it would match the north-referenced XYZ coordinates from GNSS?

Any help would be highly appreciated!

XW-HKU, 2021-07-12:
Actually integration with gnss through ekf is easy but you must be care about the pos measurement quality. Does your gps have pos covariance output?
For initial yaw, i think there must be a initiallize procedure to estimate the extrinsic between the gps frame and fastlio's global frame (the first imu frame)

salovision, 2021-07-12:
> Actually integration with gnss through ekf is easy but you must be care about the pos measurement quality. Does your gps have pos covariance output?

Yes, pos covariance is known.  I was thinking to use it anyway only when centimeter level precision can be achieved.  I use GNSS RTK so position accuracy is good.

Any tips on how to do the integration with IKFoM?

> For initial yaw, i think there must be a initiallize procedure to estimate the extrinsic between the gps frame and fastlio's global frame (the first imu frame)

Yeah, I was thinking the same, what's the right place to adjust the fastlio's global frame?  Let's say I have some initialization procedure in place and know the heading direction, what's the correct place to set that intial direction?

Thanks.

XW-HKU, 2021-07-12:
I think the integration of gnss in kalman filter will be straight forward. Just add one more terms in the measurement equation, considering the pose covariance. But you may need to fully understand the kalman filter before development.

Actually we are working on the integration of gnss now. You can wait for some time to see it.

获取 Outlook for Android<https://aka.ms/ghei36>

XW-HKU, 2021-07-12:
I think the integration of gnss in kalman filter is pretty straight forward. Just add one more terms in the measurement equation, considering the pose covariance. But you may need to fully understand the kalman filter before development.

Actually we are working on the integration of gnss now. You can wait for some time to see it.

salovision, 2021-07-12:
Excellent, looking forward to it!

closed 2021-07-12 by salovision

h-rover, 2021-08-10:
Hi,

Can we know how long the integration of GNSS gonna take ? Thanks.

insistyxj, 2022-05-31:
@XW-HKU how to fuse the gnss, i know to modified the H matrix, can you paste your code here?

chengwei0427, 2022-07-19:
> I think the integration of gnss in kalman filter is pretty straight forward. Just add one more terms in the measurement equation, considering the pose covariance. But you may need to fully [...]

Any new developments about the integration of gnss?

edwardes2201, 2023-11-11:
HELLO FRIEND, DID YOU ACHIEVE THE INTEGRATION OF GNNSS AND FASTLIO? GREETINGS.

rahulsharma11, 2024-01-17:
HELLO FRIEND, ANY UPDATE?

cc1630, 2024-06-04:
Hello, is there something new about integration of gnss? Thanks.