hku-mars/FAST_LIO #38: RViz performance

Jaetriel, 2021-07-12:
Hello, thank you again for this amazing project, I've been seeing pretty good results using it so far. One small issue I've had is in RViz when visualizing the data, I usually get 3-10 fps only and it works slowly even though the map looks great, I think it's just a bit too many points for my machine to handle. I'm running on an NVidia Jetson Xavier with an OS1-128, any suggestions about maybe somehow reducing the number of points RViz visualizes so I can improve performance a bit?

XW-HKU, 2021-07-12:
Yes, the rviz will influence the runtime performance, as the discussion in #30. You can set the parameters dense_publish_enable to 0 to low down the number of points shown. Or if you don't have to see the point cloud during the works in Xavier, you can also set the scan_publish_enable to 0 to disable all the points publish. 

And actually rviz is a monster eating the computing resources even there is nothing to show. So keep the rviz closed in the low-power platforms like ARM based computers.

Jaetriel, 2021-07-12:
@XW-HKU thank you, I will try those changes!

XW-HKU, 2021-07-13:
Would you mind update the runtime performance after you change the parameters? Thank you very much!

Jaetriel, 2021-07-13:
@XW-HKU so, disabling the publishing of the whole scan improves it, but since I want to see some of the scan at least, I tried `dense_publish_enable` set to 0 which seems to work pretty well and takes up a lot less RAM now and doesn't stress the CPU as much. The thing I found most weird was I had only 10 fps in rviz which I now saw you set from the rviz_cfg file, I guess you did it because the map is only pusblished with 10 hz? But I changed that to higher just to make it a bit smoother to look around the cloud in rviz

XW-HKU, 2021-07-13:
the rviz_config is just set casually. fastlio's pointcloud update frequency is totally decided by the input lidar data. Rviz itself will low down the render fps if the accumulated point is too large.

XW-HKU, 2021-07-13:
> Thank you, I set the frame rate for the frame to 60 for rviz and without the dense point cloud published, visualization in rviz seems a lot better on my machine, and just to confirm from what I saw in the code the PCD map saved is always from the dense point cloud even if `dense_publish_enable` is set to false right?

yes，i thought pcd may need all the point. You can change the code to what you want.

closed 2021-07-13 by XW-HKU