hku-mars/FAST_LIO #39: Support for L515

satyajitghana, 2021-07-14:
Hey, do you guys have any pointers on how to integrate L515 with FAST_LIO

I've seen https://github.com/hku-mars/FAST_LIO/blob/c8aea0dd95e4f78203b79329dd59ccdd687c1bd7/src/preprocess.cpp#L49-L66

And also seen the specific handers for each lidar, but i am not able to figure out how to write a `l515_handler`, in specific i can't figure out the processing function to convert the PCL data from L515 into `PointCloudXYZI` format.

XW-HKU, 2021-07-15:
integrate l515 is a little bit complicate.

First, if you can get the pointcloud2 topic same as velodyne,then you do not need to modify preprocess.cpp.

Besides, since the l515's a lidar frame is sampled at a same time, not like velodyne that sampled as a sequence of time, So you need to close the backward propagation in imu_processing.cpp

In the end, you should set the filter_size_surf and filter_size_map to smaller ones.

gongyue666, 2021-11-01:
@Leonard-Yao can it work with L515? I am trying to do the same thing .

XW-HKU, 2021-11-01:
> @Leonard-Yao can it work with L515? I am trying to do the same thing .

the support for L515 will soon be released

Leonard-Yao, 2021-11-02:
> > @Leonard-Yao can it work with L515? I am trying to do the same thing .
> 
> the support for L515 will soon be released

@XW-HKU cheers and looking forward to. Please let me know if it`s released, thanks advance.

Leonard-Yao, 2021-11-02:
@gongyue666 It can work, but not stable.

Taeyoung96, 2022-05-25:
@Leonard-Yao   
How could you set the launch file when you use realsense l515?  
Do you have any tips?

closed 2023-01-01 by stale[bot]