hku-mars/FAST_LIO #40: Large drift in generated map using Velodyne VLP 16

aditdoshi333, 2021-07-16:
Hello,

Thanks for sharing such amazing work. I am using fastlio 2 for map generation using Velodyne vlp16 and intel l515 (for imu). For some scenarios, it is generating an amazing map. But for some, there is a large drift during mapping. Can you please help me if there is any fine-tuning needed for the same?

Snapshot (Scene: Car parking basement)
[image]

XW-HKU, 2021-07-16:
Thanks for your tests. I may need more information about this abnormal drift:

1. Was the vlp16 and L515 time synced?
2. Is the extrinsic between imu and lidar set right? I recommend to let their axis orientations be the same.
3. Was there any message drop in the data sequence? You can rostopic echo imu_topic/header to see if the timestamp are normal when when the drift happens.

chengwei0427, 2021-07-17:
hi @aditdoshi333 ，I have a similar problem, after a thorough investigation, I found that there was no IMU data for some time.You can take a test and check it out.

XW-HKU, 2021-07-17:
> hi @aditdoshi333 ，I have a similar problem, after a thorough investigation, I found that there was no IMU data for some time.You can take a test and check it out.

You are right, if imu drops for while, fastlio may fail, it is truely a problem we will fix in the future.

aditdoshi333, 2021-07-17:
@XW-HKU @chengwei0427 Thanks for the update. I will check and update..

closed 2023-01-01 by stale[bot]