hku-mars/FAST_LIO #41: Can it be used without IMU / IMU recomendations ?

Mikor-mkr, 2021-07-19:
Hello everyone, thanks for taking the time to answer.

Is it possible to run without IMU because my IMU is unreliable? Also, can somebody recommend any budget IMU ? Thanks in advance. My lidar is a VLP-16.

XW-HKU, 2021-07-19:
IMU is important for the point-cloud un-distortion. Removing IMU will largely low down the performances of LiDAR SLAM.

For VLP-16, IMUs such as MicroStrain 3DM-GX5-25, MTI-300, MTI-680G should work, but you need to achieve the time sync.

Mikor-mkr, 2021-07-19:
@XW-HKU Thank your for your fast response, I will check the IMUs, is it possible to do the synchronization using ros packages?

closed 2021-07-19 by Mikor-mkr

XW-HKU, 2021-07-19:
> @XW-HKU Thank your for your fast response, I will check the IMUs. I close the issue as the main questions are answered but is it possible to do the synchronization using ros packages?

I strongly recommend to use hardware sync, not the software synchronization.