hku-mars/FAST_LIO #44: Time Synchronization

LeisureLei, 2021-07-20:
Hi @XW-HKU 
"Please make sure the IMU and LiDAR are Synchronized, that's important."
***
Does it mean to ensure that the time source between LiDAR and IMU is the same?

XW-HKU, 2021-07-21:
That means lidar and imu share a same time origin (or same time base).

LeisureLei, 2021-07-21:
> That means lidar and imu share a same time origin (or same time base).

Thanks!

closed 2021-07-21 by LeisureLei

h-rover, 2021-07-27:
How to achieve this external time-synchronization between an external IMU and Livox Mid-70 ? Thanks.

XW-HKU, 2021-07-28:
> How to achieve this external time-synchronization between an external IMU and Livox Mid-70 ? Thanks.

Mid 70 support three time synchronization signal types: IEEE 1588-2008 (PTPv2), PPS (Pulse Per Second), GPS (PPS+UTC). Please refer to [manual](https://terra-1-g.djicdn.com/65c028cd298f4669a7f0e40e50ba1131/Download/Mid-70/new/Livox%20Mid-70%20User%20Manual_EN_v1.2.pdf) or [specs](https://www.livoxtech.com/mid-70/specs) of mid 70 for more informations. Your IMU should produce one of these three signals to make time synchronization work.