hku-mars/FAST_LIO #45: failed to find match for field 'time' when use velodyne-16

sihanh, 2021-07-22:
the slam system has the above error when running and the map built has wrong transform when the robot itself rotates, nothing wrong when running in straight line

XW-HKU, 2021-07-22:
> the slam system has the above error when running and the map built has wrong transform when the robot itself rotates, nothing wrong when running in straight line

1. is the extrinsic set right (please refer to section in readme)？
2. is your velodyne rostopic has each point’s timestamp？

sihanh, 2021-07-23:
Thanks for answering. Extrinsic is correct since I have imu and lidar same frame direction. The topics content from Velodyne has only timstamp info in the header, as for the timestamp of each point, I guess no.

XW-HKU, 2021-07-24:
> > > the slam system has the above error when running and the map built has wrong transform when the robot itself rotates, nothing wrong when running in straight line
> > 
> > 
> > 
> > 1. is the extrinsic set right (please refer to section 3.3 in readme)？
> > 2. is your velodyne rostopic has each point’s timestamp？
> 
> Thanks for answering. Extrinsic is correct since I have imu and lidar same frame direction. The topics content from Velodyne has only timstamp info in the header, as for the timestamp of each point, I guess no.

I recommend using the dataset where each point has its own timestamp. The point cloud 2 rostopic type support it originally.

jaychiou123, 2022-10-20:
Hey @sihanh 
Have you already solved this issue?
Could you please share how you tackled this?
Thanks

nvhungv2k, 2024-06-25:
> the slam system has the above error when running and the map built has wrong transform when the robot itself rotates, nothing wrong when running in straight line

I think that this is only warning.
I run FAST-LIO with [kr_autonomous_flight](https://github.com/KumarRobotics/kr_autonomous_flight). This warning is still existed but FAST-LIO Odom is quitly good. View the results [here](https://youtu.be/siLgkC6z1NM)

txq23, 2025-08-04:
> 您可以尝试使用最新的LiDAR驱动程序（我的激光雷达是leishenC32），我通过这种方式解决了“无法找到匹配的现场时间”的问题。

I'm having the same Could you provide a link to the latest driver?issue. I'm also using the Leishen radar, but I don't have the latest driver here. Where can I download it?

lijing137, 2025-08-14:
> > 您可以尝试使用最新的LiDAR驱动程序（我的激光雷达是leishenC32），我通过这种方式解决了“无法找到匹配的现场时间”的问题。
> 
> I'm having the same Could you provide a link to the latest driver?issue. I'm also using the Leishen radar, but I don't have the latest driver here. Where can I download it?

You can leave a message on leishen’s official website, or look for their contact information.