hku-mars/FAST_LIO #46: Question about the cpu consumption

chengwei0427, 2021-07-22:
Hi,Dr Xu
Thank you for the amazing work，I have a problem about the program cpu.
**my computer is Ubuntu16.04 : intel core-i7-10700F@2.9GHz*6,GeForce GTX 1660Ti/PCle/SSE2**
I have a **vlp16**, and test the data I recorded use the **mapping_velodyne.launch** with the **default parameters**,
I find that, the data processing frequency is lower than 10Hz[**Not real time**] in the first few minutes, and the CPU usage is also high.
Is this normal?
[image]

thanks.

XW-HKU, 2021-07-22:
That's really abnormal, could you share me a piece of data?

XW-HKU, 2021-07-22:
What happens if you set scan_publish_en in velodyne.yaml to 0?

chengwei0427, 2021-07-23:
Hi,Dr Xu
Thanks for your reply.
I test a new data set today, the cpu is still very high(about 160%).
`What happens if you set scan_publish_en in velodyne.yaml to 0?`
The cpu is about 150% after set scan_publish_en and dense_publish_enable to 0.
`And you can try the avia datas in the readme to check the processing time for a avia scan. `
I can't access Google because of the gfw.
Here is my test dataset.
链接：https://pan.baidu.com/s/1epdYQ7_PmyQiVeD4Iw9iew 
提取码：b4qt

thanks!

XW-HKU, 2021-07-23:
> Hi,Dr Xu
> Thanks for your reply.
> I test a new data set today, the cpu is still very high(about 160%).
> `What happens if you set scan_publish_en in velodyne.yaml to 0?`
> The cpu is about 150% after set [...]

I am downloading the bag.

Maybe the cpu consume rate shown in **top** commend is not important since 100% stands for using only one core. You can use **sudo htop** to check each CPU core's usages. The more important thing is the real-time performence, the easist way to varify it is stopping the bag play too see if the fast-lio update is stop in the same time.

XW-HKU, 2021-07-23:
> Hi,Dr Xu
> Thanks for your reply.
> I test a new data set today, the cpu is still very high(about 160%).
> `What happens if you set scan_publish_en in velodyne.yaml to 0?`
> The cpu is about 150% after set scan_publish_en and dense_publish_enable to 0.
> `And you can try the avia datas in the readme to check the processing time for a avia scan. `
> I can't access Google because of the gfw.
> [image]
> 
> [image]
> 
> thanks!

I already downloaded it.

And seems it consumes smaller than 10 ms in my computer (intel i9-10980HK CPU @ 2.40GHz, 16G RAM). The ave total stands for the average processing time for a LiDAR scan.

[image]

chengwei0427, 2021-07-23:
Hi,Dr Xu
Thanks for your reply.
It's **x86_64** after the commend **uname -p**;

XW-HKU, 2021-07-23:
> Hi,Dr Xu
> Thanks for your reply.
> It's **x86_64** after the commend **uname -p**;
> the dependency is Eigen:3.2.92,pcl:1.9
> **Maybe something wrong with my dependency?**
> [image]
> 
> [image]

May be you can try to upgrade eigen to 3.3. looks like the processing time increase is relative to the matrix computation.

chengwei0427, 2021-07-23:
Hi,Dr Xu
The cpu consume is about 45% after update Eigen version[Eigen 3.3.6],Is this normal?
[image]
Thank you very much.

XW-HKU, 2021-07-23:
That looks very normal, and how about the processing time?

chengwei0427, 2021-07-23:
The average processing time is around 9ms, it's amazing!
[image]
[image]

XW-HKU, 2021-07-23:
Thanks, that's great

chengwei0427, 2021-07-23:
Thank you for your patient reply,
 I close the issue as the questions are solved.

Thanks again

closed 2021-07-23 by chengwei0427