hku-mars/FAST_LIO #47: Relocalization with Saved PCD Map

David9696, 2021-07-24:
Hi,

I have used this package to create a very good 3D map with Livox Mid lidar. With the saved PCD file, is it possible to use this package to do relocalization? Something like the following package:

https://github.com/Livox-SDK/livox_relocalization

Thanks for your great package.

David.

kxch928925819, 2021-07-24:
我也在琢磨这个！1、如果Evo比较熟悉，可以试试 2、可以采用cloudcopare ，手动移 
水平有限，思路而已！

 

 
 
https://github.com/Livox-SDK/livox_relocalization

HViktorTsoi, 2021-08-09:
> Hi,
> 
> I have used this package to create a very good 3D map with Livox Mid lidar. With the saved PCD file, is it possible to use this package to do relocalization? Something like the following package:
> 
> https://github.com/Livox-SDK/livox_relocalization
> 
> Thanks for your great package.
> 
> David.

Hi, we've implemented a simple relocalization system based on FAST_LIO, please take a look! https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION

XW-HKU, 2021-08-09:
> Hi, we've implemented a simple relocalization system based on FAST_LIO, please take a look! https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION

Thank you very much! Your works are great!

h-rover, 2021-08-09:
@HViktorTsoi Thanks for letting me know. Great work!

David9696, 2021-08-16:
Thanks. It works like a charm.

closed 2021-08-16 by David9696