hku-mars/FAST_LIO #52: Odometry Drifted with error Integer indices would overflow

David9696, 2021-07-30:
Hi,

When I tested this package with Mid70 Lidar, I got the following error:

Leaf size is too small for the input dataset. Integer indices would overflow.[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.

Then, the published odometry would drift to very far distance. Is this normal?

Thanks.

XW-HKU, 2021-07-30:
Could you produce more infomation like the screen shot of the point cloud when this error shown.

David9696, 2021-07-30:
Hi,

Thanks for your reply. I have set the param of scan_publish_en to 0. Hence, no point cloud is published. This problem appears when the lidar was not moving for more than 45 minutes. Then, the odometry drifted to very far distance. Is this behaviour expected?

Thanks.

David

XW-HKU, 2021-07-30:
Weather if the pcd_save_function is ON?
Weather if the odometry is fine before large drift happens?

David9696, 2021-07-31:
Hi,

The pcd_save_function has been turned off.

Odometry was totally fine  before the large drift.

David9696, 2021-08-04:
Hi,

Quite confirm the problem can be reproduced if the lidar is not moving for more than 45 minutes. Once it moves after 45 minutes, the node would crash.

Hope this info helps.

David9696, 2021-08-16:
Hi, 

We found out that the Odmetry is drifted exponentially when the lidar is facing a wall and not moving for long period of time. Is it possible to halt the Odometry estimation when the lidar is not moving? Would it solve the problem?

XW-HKU, 2021-08-16:
> Hi,
> 
> We found out that the Odmetry is drifted exponentially when the lidar is facing a wall and not moving for long period of time. Is it possible to halt the Odometry estimation when the lidar is not moving? Would it solve the problem?

Of course you can disable the update when degeneration happens but you cannot know weather if the LiDAR is moving with constant velocity or totally not moving, only through LiDAR and IMU inputs.

XW-HKU, 2021-08-16:
> 5 minutes

Sorry to missing this issue, I tested some dataset with over 1 hour bag but things are normal. I wonder what kind of crash you mean? the large drift or error report in the terminator?

XW-HKU, 2021-08-16:
> Hi,
> 
> Quite confirm the problem can be reproduced if the lidar is not moving for more than 45 minutes. Once it moves after 45 minutes, the node would crash.
> 
> Hope this info helps.

Sorry to missing this issue, I tested some dataset with over 1 hour bag but things are normal. I wonder what kind of crash you mean? the large drift or error report in the terminator?

David9696, 2021-08-16:
Hi,

No problem. I am just trying to see whether I can help to identify the problem. The error seems not consistent. There was a time that the odometry grew exponentially after the lidar was not moving for two to three hours. In another test, the node was not crashed but the Odometry topic was empty  after the lidar was not moving for three to four hours. Also, there was a test in which the node died by itself after the lidar was not moving for about an hour. For all the tests, the lidar was placed in a small room and facing a wall.

Could it be related to the IMU that I used? FYI, I am using a PX4 imu.

Thanks for your great help.

XW-HKU, 2021-08-16:
I think maybe the crash is related with that the path size is too large (especially when duration is too long) to make itself crash. So I just updated the code, added a switch (**publish / path_en** in .yaml files), when it is set to **false**, the path record and publish will be disabled. I think this may help the mentioned issues.

David9696, 2021-08-19:
Hi,

Thanks for the update. After the update, indeed the node would not crash again when running for long times.

During today testing, the estimated Odometry suddenly increase to very big value as follows:

header: 
  seq: 54331
  stamp: 
    secs: 1629366638
    nsecs: 298263311
  frame_id: "camera_init"
child_frame_id: "body"
pose: 
  pose: 
    position: 
      x: -10058.3644121
      y: -1204.08801205
      z: -17130.9135003
    orientation: 
      x: 0.006549194477
      y: 0.281932660375
      z: -0.378108663831
      w: -0.88176239509
  covariance: [1.9440237935678447e-06, -8.8770845446568e-07, 2.3484943832686953e-07, -1.4186796664153743e-05, 1.3940757544851185e-05, 1.187292981730975e-05, -8.877084544709174e-07, 7.304837791494663e-07, -5.3140617177034306e-08, -2.8776159334385457e-06, -1.2507521438026115e-05, 3.5104065157404486e-06, 2.3484943832768935e-07, -5.3140617180252396e-08, 1.7477052605659665e-07, -3.500118707457511e-07, -6.9677832176236534e-06, 2.730571495773008e-06, -1.4186796664217184e-05, -2.8776159335039866e-06, -3.500118706518139e-07, 0.0004909961551499604, -1.698745922613648e-05, -0.000334378944620366, 1.3940757544920728e-05, -1.2507521437914429e-05, -6.967783217672056e-06, -1.6987459226105017e-05, 0.0007855388681740228, -7.309354893071643e-05, 1.1872929817260727e-05, 3.510406515692055e-06, 2.7305714958239888e-06, -0.0003343789446203961, -7.309354893079254e-05, 0.00034836881849878627]

No error message from the node. May I know, under what situation would the estimated /Odometry increase exponentially like this?

FYI, the livox lidar was moving when the Odometry was increasing exponentially.

XW-HKU, 2021-08-19:
> Hi,
>
> Thanks for the update. After the update, indeed the node would not crash again when running for long times.
>
> During today testing, the estimated Odometry suddenly increase to very big value as [...]

Did this still happened after a long-time stand by?

David9696, 2021-08-19:
Hi,

No. This happened when I was moving the lidar and building a map. The lidar was moving up at that instant.

Do you need any debug information?

XW-HKU, 2021-08-19:
> Hi,
> 
> No. This happened when I was moving the lidar and building a map. The lidar was moving up at that instant.
> 
> Do you need any debug information?

That‘s quite abnormal, did you checked if your extrinsic is right? and the time sync?

David9696, 2021-08-20:
Hi,

Yes, the extrinsic setting is correct and the time is in sync. I have successfully built a few maps before by using this package.

My livox lidar and imu are tilted up by 30 degrees during mapping. Could it be the reason?

FYI, I managed to build a few maps successfully before with the tilted up  lidar.

David9696, 2021-08-23:
Hi,

I realised that when there is "lidar loop back, clear buffer" error, the Odemetry starts to drift exponentially.

May I know why there is a lidar loop back?

I have another machine with the same Livox lidar. But I have never seen this error.

Hope you could enlighten me.

Thanks.

XW-HKU, 2021-08-23:
> Hi,
> 
> I realised that when there is "lidar loop back, clear buffer" error, the Odemetry starts to drift exponentially.
> 
> May I know why there is a lidar loop back?
> 
> I have another machine with the same Livox lidar. But I have never seen this error.
> 
> Hope you could enlighten me.
> 
> Thanks.

Please check your data quality, this error happend may because of the data package lost.

David9696, 2021-08-23:
Hi,

Do you mean the lidar is malfunction? Is there a way to check the data quality?

XW-HKU, 2021-08-23:
LiDAR or IMU package lost.

Here is a way to check:

after the drift happend, terminate the code. then

```
cd src/FAST_LIO/Log
python plot.py
```

check if there is any irregulations int the state estimation figures.

David9696, 2021-08-23:
Hi,

I set "runtime_pos_log_enable" to 1 and manage to get the log files for os_log.txt,mat_pre.txt and map_out.txt .

However, the imu.txt is empty which results in the following error when I tried to plot.

plot.py:37: UserWarning: loadtxt: Empty input file: "imu.txt"
  imu=np.loadtxt('imu.txt')
Traceback (most recent call last):
  File "plot.py", line 38, in <module>
    time=imu[:,0]
IndexError: too many indices for array

Do you know why?

XW-HKU, 2021-08-23:
did u update to latest code

David9696, 2021-08-24:
Hi,

Yes, I am using the latest code. Downloaded via git clone. However, the imu.txt is remained blank. Is it because my imu has issue?

XW-HKU, 2021-08-24:
> Hi,
> 
> Yes, I am using the latest code. Downloaded via git clone. However, the imu.txt is remained blank. Is it because my imu has issue?

just comment the line 38 in plot.py.

David9696, 2021-08-24:
Hi,

Thanks. I have updated to the latest code. This is the picture I have. I stopped the mapping right after I saw the lidar loop back error. How could like read this picture?

[image]

David9696, 2021-08-24:
Hi,

This is another picture when Odometry started to drift exponentially.

[image]

XW-HKU, 2021-08-24:
> Hi,
> 
> This is another picture when Odometry started to drift exponentially.
> 
> [image]

you can zoom the abnormal part to see if there is any data jump or data loss.

David9696, 2021-08-24:
Hi ,

After zooming the picture, I realised that is the out-y and pre-y velocity was increasing suddenly to a very big value(more than 9.0). May I know what does it mean?

XW-HKU, 2021-08-25:
_lidar loop back, clear buffer_ This error happens mostly when the timestamp of LiDAR jumps back.

David9696, 2021-08-25:
Hi,

Do you know why there is a jumps back in timestamp for the lidar? How to ensure the jump back would not affect the mapping accuracy? Whenever there is a jump back, the fast-lio mapping process freezes for a while and then odometry drifts to a far distance.

FYI, when livox_mapping package is in used, the mapping process would be continuous even when there is a jump back.

WeiyiKong-0419, 2023-04-20:
> Hi,
>
> Do you know why there is a jumps back in timestamp for the lidar? How to ensure the jump back would not affect the mapping accuracy? Whenever there is a jump back, the fast-lio mapping process [...]

Have you solved this question? I have also met the loopback situation. And the map will freeze and drift.

wyb17js, 2023-10-25:
hi,I have encountered the same problem as you, the positioning is lost after running for a while, the car drifts, I would like to know how you solve this problem
thank you!

wyb17js, 2023-10-26:
> WeiyiKong
Hi,
have you solve your question?I experienced odometer drift after a period of successful positioning.I hope your experience can help me.

Kmedrano101, 2025-03-31:
Hi @David9696 were you able to fixed the issue? I am facing something similar, exponential drifting