hku-mars/FAST_LIO #53: Real-time performance

Jaetriel, 2021-07-30:
Hey, I noticed in FAST-LIO sometimes when I play a ROS bag with normal speed, the bag finished playing and everything is okay, but it takes FAST-LIO more time and it keeps processing the data even after the bag finishes playing, is this normal behaviour?

XW-HKU, 2021-07-30:
I think we discussed it in #38 and #43.

And I have a suggestion, when talking about the computational efficiency, please produce enough infomation such as:**your configuration (if publish every point, if save pcd)**,  **cpu information**, **LiDAR type**, **if the dependencies are fulfilled**.

Jaetriel, 2021-07-30:
Yes you are right we discussed performance but I don't think this aspect specifically and I wasn't sure if it was cause by rviz exactly. But my configuration is:
dense_pub_en: false
save_pcd:true

using OS1-128 lidar with an NVidia Jetson Xavier with 8core arm64 processor

XW-HKU, 2021-07-30:
what happend if you close the save_pcd? will it be real-time?

chengwei0427, 2021-07-31:
Hi, You can have a look #46   @Jaetriel

Jaetriel, 2021-08-25:
Hi, I'm sorry I've been away for quite some time. I tested a bit today, putting save_pcd to false helped a tiny bit, but still nowhere near real-time, my lidar spins at 10hz speed but when I checked the /cloud_registered topic it's being published with 5-6hz sometimes even 2-3. Then I started increasing `point_filter_num` parameter and also the filter leaf size and I started seeing improvements and got close to real-time, so I'm guessing the main issue is just the number of points with the 128 channel lidar since I get like 2.1 mil points per second?

Jaetriel, 2021-09-08:
@XW-HKU ok, thanks for the clarification. So is it correct to assume that when you have a much larger number of points publishing of the point cloud map slows down and becomes less than real-time? Also, another question I have is would it be possible at all to have a case where the sensors are placed still for longer period of time lets say a few minutes and the algorithm doesn't keep collecting the same points over and over again to fill up the ram and other resources and doesn't keep adding the same frames again and again to the pcd map?

closed 2023-01-01 by stale[bot]