hku-mars/FAST_LIO #54: Do you optimize the extrinsic between lidar and imu?

CanCanZeng, 2021-08-02:
Hi, I read your paper about fast-lio2, you say that the dimension of measurement is 24, including extrinsics between lidar and imu. But in the code you set the measurement dimentsion to 18 https://github.com/hku-mars/FAST_LIO/blob/f159d57da8fa48920cefae1e9b289cb1785e4faa/include/common_lib.h#L21, does that mean you did not optimize the extrinsic?

XW-HKU, 2021-08-02:
The extrinsic is optimized in fastlio2, but the performance depends on the richness of excitation. 

DIM_STATE is no longer used in the state estimation of FAST-LIO2.

CanCanZeng, 2021-08-03:
Thank you for your reply! Then where can I output the optimized extrinsic? The point cloud is not very clean and accurate for my own data, I suspect this ascribes to my inaccurate extrinsic.

CanCanZeng, 2021-08-03:
O.K. I think I find it, it is defined here https://github.com/hku-mars/FAST_LIO/blob/f159d57da8fa48920cefae1e9b289cb1785e4faa/include/use-ikfom.hpp#L15

CanCanZeng, 2021-08-03:
In my indoor data, I find that the translation between lidar and imu is keep increasing, this seems not expected, so is there any way to disable optimize extrinsics after some time?

XW-HKU, 2021-08-09:
> In my indoor data, I find that the translation between lidar and imu is keep increasing, this seems not expected, so is there any way to disable optimize extrinsics after some time?

Since the performance of extrinsic estimate depends on the motion aggressiveness, the more aggressive the better. I recommend you shake the lidar for a while during the data capture, than see how the extrinsic estimates change. If still diverges, maybe you should change the extrinsic settings

CanCanZeng, 2021-08-09:
I'm sorry I didn't make it clear. The optimized extrinsic is keep incresing, but still within an acceptable range in about 10min data length. I think the extrinsic is a bit large than expected but it still not diverges, so I think the better way is to fix the extrinsic after some time or when the uncertainty is below to some degree.
Like in vins-mono, there is a flag to enable extrinsic optimization, so realize the above strategy is rather simple. I was wondering if fast-lio2 can do similar thing, but tried in vain finding such flag.

XW-HKU, 2021-08-09:
> I'm sorry I didn't make it clear. The optimized extrinsic is keep incresing, but still within an acceptable range in about 10min data length. I think the extrinsic is a bit large than expected but [...]

I think you are right. I will add this function right now.

XW-HKU, 2021-08-09:
I added the switch for the online extrinsic estimation in the yaml file, please check it, thanks.

CanCanZeng, 2021-08-09:
Thank you, I think it's enough for my purpose. I can print out the optimized extrinsic and then write it in config file, then disable online optimization.

closed 2021-08-13 by CanCanZeng