hku-mars/FAST_LIO #56: Qvector error

kxch928925819, 2021-08-12:
.. logging to /home/dji/.ros/log/c0ec4d26-fb32-11eb-b0bf-34d26269e60d/roslaunch-dji-MANIFOLD-2-C-5851.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dji-MANIFOLD-2-C:42909/

SUMMARY
========

PARAMETERS
 * /common/imu_topic: /livox/imu
 * /common/lid_topic: /livox/lidar
 * /common/time_sync_en: False
 * /cube_side_length: 1000.0
 * /dense_publish_enable: True
 * /feature_extract_enable: False
 * /filter_size_map: 0.5
 * /filter_size_surf: 0.5
 * /mapping/acc_cov: 0.1
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 260.0
 * /mapping/extrinsic_R: [1, 0, 0, 0, 1, 0...
 * /mapping/extrinsic_T: [0.05512, 0.02226...
 * /mapping/extrinsic_est_en: True
 * /mapping/fov_degree: 100
 * /mapping/gyr_cov: 0.1
 * /pcd_save_enable: False
 * /point_filter_num: 3
 * /preprocess/blind: 4
 * /preprocess/lidar_type: 1
 * /preprocess/scan_line: 6
 * /publish/dense_publish_en: True
 * /publish/scan_bodyframe_pub_en: True
 * /publish/scan_publish_en: True
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /runtime_pos_log_enable: False

NODES
  /
    laserMapping (fast_lio/fastlio_mapping)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[laserMapping-1]: started with pid [5866]
process[rviz-2]: started with pid [5867]
Multi thread started 
p_pre->lidar_type 1
~~~~/home/dji/catkin_ws/src/FAST_LIO/ file opened
[ INFO] [1628748102.704890707]: IMU Initial Done
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)

XW-HKU, 2021-08-12:
what happens?

closed 2021-08-13 by kxch928925819

WadeFrank, 2022-04-19:
我遇到了同样的问题，你是怎么样解决的啊？ @kxch928925819

x710777335, 2023-12-14:
你好，我遇到的同样的问题，请问你解决了吗？

wuming122333, 2024-02-24:
> 你好，我遇到的同样的问题，请问你解决了吗？
您好，我也遇到了这个问题，请问您如何解决的，谢谢！！！