hku-mars/FAST_LIO #57: Flyaway/Drift on Drone

TheRealKaufmann, 2021-08-12:
Hi there,

we are experimenting with FAST-LIO 2 on a custom quadcopter. We use an Ouster OS1-32 Lidar with the built in IMU and configured everything accordingly. When just walking around with the Ouster itself, everything works perfectly. But when putting it on a drone, we often encounter "flyaways" and drift. 

We suspect two causes:
- Vibration on the IMU
- Downwards field of view is blocked by spinning propellers

Now my question: How much impact should these two actually have on the FAST-LIO algorithm? What can be done configuration wise to make the system more reliable?

Here is a picture of the drone:
[image]

I hope you can help out here.

Thanks,
Tom

XW-HKU, 2021-08-12:
I may need more information, such as how does the environment like (photos or rviz screenshots)? 

What happens if you shake the LiDAR on the hands?

TheRealKaufmann, 2021-08-13:
Thank you very much for your answer. Vibration is quite a lot I think, but we also tried physically dampening the lidar. 
I will gather flight data of the environment next week. Also, when flying indoors (eg. a simple room), the localization does not fail. But when trying outdoors, it fails quite often. Without the drone flying and just using the lidar itself, everything works fine outdoors too. But, as I said, I will gather data next week and upload it.

Until then, thank you very much for your help so far. You will hear from me next week.

closed 2021-08-30 by TheRealKaufmann

CrypticInfusion, 2025-03-19:
This isn't directly related to your issue, but I'm working on a project that requires the Ouster OS1-32, and I was curious about your setup process. I've been trying to get FAST-LIO to work with mine, but I keep running into issues. Any insights or guidance would be greatly appreciated!

sanjana-dev9, 2025-04-28:
What's the speed of the drone?
Did you get it working with Internal IMU data?
Did you store the lidar point in PointCloud2 or CustomMsg format while capturing the data?

CrypticInfusion, 2025-04-30:
> What's the speed of the drone? Did you get it working with Internal IMU data? Did you store the lidar point in PointCloud2 or CustomMsg format while capturing the data?

Actually, this was for my senior design project which was focused on an indoor mapping robot, a ground robot rather than a drone. I did manage to get it working with the Internal IMU and I stored the lidar point in PointCloud2, but I had a really bad drifting problem. The system was relatively stable as long as the OS1 remained stationary, but as soon as it moved, the odometry became extremely unstable. After a lot of troubleshooting, I suspected the problem was due to some weird timing offset between the lidar and IMU, but I wasn’t able to fully resolve it.

sanjana-dev9, 2025-05-01:
> Actually, this was for my senior design project which was focused on an indoor mapping robot, a ground robot rather than a drone. I did manage to get it working with the Internal IMU and I stored [...]

Even I am facing similar issues when my drone is stationary vs moving. Even though there are good amount of overlap between the lidar frames.
I am using Livox avia and storing the data in PointCloud2 format and then converting to CustomMsg (storing the data in CustomMsg is really clogging up rest of the processes) to pass onto FAST_LIO.

My questions were directly towards @TheRealKaufmann and his setup.

> Hi there again,
> 
> sorry for the late reply. We have been recording bags outdoor on a parking lot with buildings. It looks like this: [image]
> 
> I was flying in manual mode and recorded the datasets in very heavy wind with strong vibrations. This is the exact location where the drone always fails to keep position. The strange thing is that on my computer all the datasets are processed perfectly by FAST-LIO2. The Lidar can even see the ground in most parts.
> 
> [image]
> 
> It is weird that it worked now. We also processed the bag on the drone itself and it still worked.
> 
> What effects has 'standing still' on the algorithm? Is it more likely to fly away if we just hover at one position and not move a lot?
> 
> Also, is FAST-LIO more likely to fail if it takes longer to process? (for example on the Jetson Xavier NX on the drone) Is there a difference between providing a bag vs live data?
> 
> We will be doing more testing and data recording in the next days.
> 
> Thanks, Tom