hku-mars/FAST_LIO #58: 建图z轴漂移问题

maomao0220, 2021-08-15:
首先非常感谢您开源这个代码，非常棒。
您好，我在用AVIA跑fast_Lio建图时，建立的地图不在栅格水平面上，估计出的雷达TF的Z轴会越来越高，然后越来越低，在跑自己的数据集时，看position的Z轴坐标最大13左右，最小负几，请问什么原因，有什么解决或缓解办法？？谢谢解答
[image]
[image]
[image]

maomao0220, 2021-08-15:
以及像这种建立出来的地图，对后面重定位和导航有很大的影响吗？？？？

XW-HKU, 2021-08-16:
这个不叫漂移，只是因为world坐标系z轴不是铅锤方向，地图本身应该是正常的。

fastlio的world坐标系等于雷达初始的body xyz轴。

maomao0220, 2021-08-16:
那您的意思是像这种建出来的地图也是可以用的吧，比如说用来导航路轨，

maomao0220, 2021-08-16:
那您的意思是像这种建出来的地图也是可以用的吧？？？比如说用来进行导航或路轨

maomao0220, 2021-08-16:
我用lili-om跑这个数据集时，就没有出现上述z轴的情况

XW-HKU, 2021-08-16:
你这个问题我没法回答，导航能不能用取决于导航那边是怎么写的

XW-HKU, 2021-08-16:
> 我用lili-om跑这个数据集时，就没有出现上述z轴

Woy01, 2021-08-16:
> > 我用lili-om跑这个数据集时，就没有出现上述z轴的情况
> 
> 那说明fastlio的坐标系定义和liliom不一样

您好，请问fast lio的地图坐标系定义是采取第一帧雷达坐标系吗？

XW-HKU, 2021-08-16:
> > > 我用lili-om跑这个数据集时，就没有出现上述z轴的情况
> > 
> > 
> > 那说明fastlio的坐标系定义和liliom不一样
> 
> 您好，请问fast lio的地图坐标系定义是采取第一帧雷达坐标系吗？

第一帧imu坐标系

closed 2021-08-19 by XW-HKU

yxllnSLAM, 2022-06-10:
@maomao0220 你好，你这个问题解决了吗？可以私聊？v16621710962

plumewind, 2022-08-12:
你好，
我发现了类似的问题。我的场景是一个下坡，可是地图是升高的，有什么改进方法或思路吗？我可以试试，比如imu初始对准？

[image]
[image]

ZYQ1919, 2023-05-22:
> 

> > > > 我用lili-om跑这个数据集时，就没有出现上述z轴的情况
> > > 
> > > 
> > > 那说明fastlio的坐标系定义和liliom不一样
> > 
> > 
> > 您好，请问fast lio的地图坐标系定义是采取第一帧雷达坐标系吗？
> 
> 第一帧imu坐标系

May I ask where this is set in the source code?

ZYQ1919, 2023-05-22:
> > > > 我用lili-om跑这个数据集时，就没有出现上述z轴的情况
> > > 
> > > 
> > > 那说明fastlio的坐标系定义和liliom不一样
> > 
> > 
> > 您好，请问fast lio的地图坐标系定义是采取第一帧雷达坐标系吗？
> 
> 第一帧imu坐标系

请问这是在源码哪里进行设置的呢？

duyingchang, 2023-09-06:
你好，请问fastlio的坐标系是雷达坐标系还是imu坐标系？