hku-mars/FAST_LIO #59: 内存占用一直上升，直到程序崩溃

oscar1780, 2021-08-16:
报错内容如下：
[laserMapping-2] process has died [pid 73469, exit code -9, cmd /home/oscar/wyz_ws/devel/lib/fast_lio/fastlio_mapping __name:=laserMapping __log:=/home/oscar/.ros/log/ebdd07f0-fe5c-11eb-aeff-19978b9088fb/laserMapping-2.log].
log file: /home/oscar/.ros/log/ebdd07f0-fe5c-11eb-aeff-19978b9088fb/laserMapping-2*.log
rosbag 包有58G, 建图过程中观察了一下内存占用情况，内存总共16G, 建图开始时有15G, 建图过程中内存一直在以0.1%每秒的速度增加，直到接近100%，然后崩溃。尝试调整过 filter因子的大小，但是不管用。
 是否有其他参数可以调整，还是单纯就是数据包太大导致崩溃，希望作者有时间可以帮忙解答一下。
[image]

chengwei0427, 2021-08-16:
如果是重复的在一个区域数据，构图过程中，内存增长到一定范围，就不怎么增长了；
很大范围的构图，还没有测试过？    @oscar1780  你的构图面积有多大啊？
不知道是不是因为map_incremental的缘故

oscar1780, 2021-08-16:
@chengwei0427  大概3200 m*m; 两条互相垂直的长走廊，长度在200m左右，宽度一个7m ,一个宽9m, 走廊两侧是镂空的，可以扫到外面的物体。刚才用小体积的bag包又跑了一遍，图可以建出来，但是CPU占用率还是一直上涨，干到80%多。
另外还想问一下，算法建出来的地图为什么这么大，同样的包用lio sam 建出来才200mb, Fast lio2 直接3个G，QAQ

chengwei0427, 2021-08-16:
cpu占用率应该还好，你想说的是内存一直增加吧？
地图大小不一样，一方面lio-sam用的是特征点，而fast-lio2用的是直接的激光数据，另一方面可能是降采样参数的关系。

XW-HKU, 2021-08-16:
> @chengwei0427 大概3200 m*m; 两条互相垂直的长走廊，长度在200m左右，宽度一个7m ,一个宽9m, 走廊两侧是镂空的，可以扫到外面的物体。刚才用小体积的bag包又跑了一遍，图可以建出来，但是CPU占用率还是一直上涨，干到80%多。
> 另外还想问一下，算法建出来的地图为什么这么大，同样的包用lio sam 建出来才200mb, Fast lio2 直接3个G，QAQ

有两个可能的原因：
1. pcd_save_en开启导致缓存中的pcd持续增长过大，最终导致内存溢出。其实pcd save那里可以自行修改成隔一段时间保存一次。
2. ros的path数量过大后会溢出，可以试试把src/laserMapping.cpp中的[599行](https://github.com/hku-mars/FAST_LIO/blob/53ee209c125ae60b75ef33bcd837f606e1f54d2f/src/laserMapping.cpp#L599)注释掉，也许就不会报错了。

XW-HKU, 2021-08-16:
我再解释下：
1. 解释下pcd save这里，不同的人需求不一样，有的人希望保存成多个pcd，有的人希望所有的点保存在一个pcd file里面，因此我pcd save这一块就采取了一个简单的逻辑：每一帧的点云都累积保存在缓存里面，最终程序退出的时候会把累积的点云一次性写到硬盘里面。这样的弊端就是：当时间过长，缓存可能会溢出导致crash。一般来讲，可以把linux的swap（虚拟内存）设置的比较大来防止溢出。
2. nav_msgs::Path这个数据结构，一旦size超过2w左右就会崩掉，如果不是一定要实时显示path，可以考虑一直把599行注释掉来防止crash。

XW-HKU, 2021-08-16:
我刚才测试了一下，可能复现了你的现象，原因就出在我说的第二点path size溢出。

另外我还测试了一下，如果path不会溢出并且关闭pcd_save_en, 内存增长大约为 30 M / 分钟 (一直往一个方向走)。

我更新了一下代码，做了两个优化：
1. 添加了path record / publish 的开关，在yaml文件中的 publish / path_en.
2. 增加了等间隔保存pcd的功能，可以设置每多少frame保存一次pcd，这样缓存里面的点云大小不会无限增加了，具体参考yaml文件里面pcd_save相关参数的注释。

oscar1780, 2021-08-17:
对，是内存一直增加

oscar1780, 2021-08-17:
> 我刚才测试了一下，可能复现了你的现象，原因就出在我说的第二点：当path size溢出后，内存会突然快速增加。
> 
> 我更新了一下代码，做了两个优化：
> 
>     1. 添加了path record / publish 的开关，在yaml文件中的 publish / path_en.
> 
>     2. 增加了等间隔保存pcd的功能，可以设置每多少frame保存一次pcd，这样缓存里面的点云大小不会无限增加了，具体参考yaml文件里面pcd_save相关参数的注释。
> 
> 
> 更新后，如果关闭path_en 和 pcd_save_en, 内存增长大约为 30 M / 分钟 (一直往一个方向走)。

oscar1780, 2021-08-19:
> Any update?

 Sorry for my late reply. Appreciate your update and detailed explanation. I've just tried my biggest rosbag (60G) and change the SWAP size up to 32GB. Now the mapping process is very stable and fluent. I've checked the pcd file in CloudCompare,well, it's very beautiful and clear compared to other methods like lio sam.
  I will close this issues and focus on the ikd-tree and IKFOM then. Thank you for your great job.

closed 2021-08-19 by oscar1780