hku-mars/FAST_LIO #61: Outdoor scan drifting

Astraing, 2021-08-20:
Hi,

I have a strange drift for indoor scan, here is my config : 
 - Livox mid-70
 - LPMS-IG1 RS232: Ultra Low-Noise 9-Axis Inertial Measurement Unit (IMU) / AHRS with USB / RS232 Connectivity
 - Livox connected by WiFi
 
For outdoor scanning, I have no problem at all, the accuracy is good.

Below the link for some screenshot and video : 
[https://drive.google.com/drive/folders/1SLiUvaxHNfvX5DnCxG2OX3Dg-SjDpjs1?usp=sharing](url)

XW-HKU, 2021-08-20:
Maybe you can try to set **filter_size_surf** and **filter_size_map** to 0.2 both.

Astraing, 2021-08-20:
Thanks ! The stability of my interior scanning improves. I have tried with filter_size_surf and filter_size_map to 0.1 both. 
I suppose that 0.2 is the minimum size in meter considered. 

But with the value sets at 0.1 and 0.2, the accuracy is not so good and it gets worse when I shake the sensor brutally.
Below the gap between two scans : 

[image]

Astraing, 2021-08-20:
I have retried the scan moving much slower and sometimes staying immobile on large surface. The accuracy is better.
Thank you for your commitment. 

[image]

closed 2021-08-20 by Astraing

85256638, 2024-11-18:
@Astraing Thanks for asking this excellent question, I am also facing the similar issue with indoor scanning. Anything that you have tweaked other than the `filter_size_surf` and `filter_size_map` to improve the indoor scan performance? Thanks in advance!