hku-mars/FAST_LIO #62: Question about the H matrix.

Gatsby23, 2021-08-23:
不好意思，想问下关于H矩阵的具体推导。我查询了R2Live中的矩阵详细Jacobian形式。但是好像和Fast-LIO第一版中的代码Jacobian推导不同？代码中的具体写法如下：
[image]
论文中的格式是：
[image]
另外一个关键的地方就是误差观测值。
我的理解是平面的误差观测方程应该是：n * (T_G_I * Point_in_lidar ) + d = 0；但我在自己实现的时候却一直误差得到的是反方向推导。觉得很奇怪，想请教下作者，这里的H和Residual具体应该怎么写？

XW-HKU, 2021-08-24:
没明白你说的反方向推导是什么。

Gatsby23, 2021-08-24:
> 没明白你说的反方向推导是什么。

谢谢您的回复，这里主要想问的是为什么代码中的公式应该是：[p_a]×R^{T}*norm_vec？和补充材料中不一样？

XW-HKU, 2021-08-24:
因为Hj是行向量，而代码里面的向量A是列向量。互相是转置关系。

Gatsby23, 2021-08-24:
> 因为Hj是行向量，而代码里面的向量A是列向量。互相是转置关系。

哦哦，谢谢，谢谢。另一个很基础的问题是关于Residual：论文里的Residual应该是投影点要减去临近点q_j，但是代码中只有投影点*平面法向量来表示在平面上，直接是point_temp * n + d = 0？

XW-HKU, 2021-08-24:
> > 因为Hj是行向量，而代码里面的向量A是列向量。互相是转置关系。
> 
> 哦哦，谢谢，谢谢。另一个很基础的问题是关于Residual：论文里的Residual应该是投影点要减去临近点q_j，但是代码中只有投影点*平面法向量来表示在平面上，直接是point_temp * n + d = 0？

和q相关的所有项，在求Jacobian的时候都消去了。

Gatsby23, 2021-08-24:
> > > 因为Hj是行向量，而代码里面的向量A是列向量。互相是转置关系。
> > 
> > 
> > 哦哦，谢谢，谢谢。另一个很基础的问题是关于Residual：论文里的Residual应该是投影点要减去临近点q_j，但是代码中只有投影点*平面法向量来表示在平面上，直接是point_temp * n + d = 0？
> 
> 首先，在求H矩阵的时候，和q相关的所有项，在求Jacobian的时候都消去了。
> 
> 其次，代码中的match部分使用了另外一个等效公式：Ax+By+Cz=d。这个公式的d是截距，（A,B,C）是法向量。这个公式可以使用小二乘更准确的求解（比u（p-q）数值精度更高）。

嗯嗯，也就是说，实际的观测是截距？用截距差来做观测更为准确对吗？

Gatsby23, 2021-08-24:
很棒的工作，多谢你们的详细讲解，为机器人论坛做出了相当棒的推进~

closed 2021-08-24 by Gatsby23

chengwei0427, 2022-07-19:
你好， @Gatsby23 

H的雅克比为
[image]
为何代码实现中，关于旋转的负号没有了？

chengwei0427, 2022-07-19:
Hi, @Gatsby23 @XW-HKU 

H的雅克比为
[image]
但是为什么在代码实现中，关于旋转的雅克比，少了一个负号呢？

chengwei0427, 2022-07-19:
> Hi, @Gatsby23 @XW-HKU
> 
> H的雅克比为 [image] 但是为什么在代码实现中，关于旋转的雅克比，少了一个负号呢？

#28 Get√
[image]

iwander-all, 2023-08-20:
I have just done an experiment on FAST-LIVO with following change:
`            // Eigen::Vector3d A(point_crossmat * mState.rot_end.transpose() * norm_vec);
            Eigen::Vector3d A = -(norm_vec.transpose() * mState.rot_end * point_crossmat).transpose();`

which satisfies the formular in paper, the code works good.