hku-mars/FAST_LIO #63: Problem with the F_x in the IMU_Porcessor file.

Gatsby23, 2021-08-24:
你好，这里我对IMU_Processor代码中的速度更新有些不理解：
在论文中的公式为：
[image]
然而在代码中：
```
    /* covariance propagation */
    Eigen::Matrix3d acc_avr_skew;
    Eigen::Matrix3d Exp_f   = Exp(angvel_avr, dt);
    // 这里是最不理解的，为什么对w_t进行反对称计算.
    acc_avr_skew<<SKEW_SYM_MATRX(angvel_avr);

    F_x.block<3,3>(0,0)  = - Exp_f;
    F_x.block<3,3>(0,9)  = - Eye3d * dt;
    // F_x.block<3,3>(3,0)  = R_imu * off_vel_skew * dt;
    F_x.block<3,3>(3,6)  = Eye3d * dt;
    // 这里为什么乘以的是旋转？
    F_x.block<3,3>(6,0)  = - R_imu * acc_avr_skew * dt;
    F_x.block<3,3>(6,12) = - R_imu * dt;
    F_x.block<3,3>(6,15) = Eye3d * dt;
```
想问下，这里是不是写错了？为什么这里的acc_avr_skew一直求的是angvel_avr的反对称矩阵？

XW-HKU, 2021-08-24:
> 你好，这里我对IMU_Processor代码中的速度更新有些不理解：
> 在论文中的公式为：
> [image]
> 然而在代码中：
>
> ```
>     /* covariance propagation */
>     Eigen::Matrix3d acc_avr_skew;
>     Eigen::Matrix3d Exp_f   = Exp(angvel_avr, dt);
>     // [...]

acc_avr_skew<<SKEW_SYM_MATRX(angvel_avr);这句话写错了，应该使用acc_avr，现在用ikfom没有这个错误了。

Gatsby23, 2021-08-24:
> acc_avr_skew<<SKEW_SYM_MATRX(angvel_avr);这句话写错了，应该使用acc_avr，现在用ikfom没有这个错误了。
OK，多谢指点，特别棒的工作，为你们点赞

closed 2021-08-24 by Gatsby23