hku-mars/FAST_LIO #67: No Effective Points!

David9696, 2021-08-30:
Hi,

May I know under what situation would the fastlio package report the following warning?

No Effective Points!

Once this warning is shown, the estimated  Odometry would grow exponentially.

Hope you could enlighten me. Thanks

XW-HKU, 2021-08-30:
> Hi,
> 
> May I know under what situation would the fastlio package report the following warning?
> 
> No Effective Points!
> 
> Once this warning is shown, the estimated Odometry would grow exponentially.
> 
> Hope you could enlighten me. Thanks

There are several conditions may lead to the  _No Effective Points!_ warning:

1. most of the points are in a very tiny space (the parameter blind controls the minimum distance fast-lio use).
2. too less points from lidar.
3. degeneration happened for a while.

David9696, 2021-08-30:
Hi,

Thanks for your enlightenment.

1. Is it meant that I have to increase the parameter blind? Currently, it is set to 0.5.
2. This error happened when I was mapping a small room with four walls. Is it what you refer as less point? Also, would decrease the parameters of "filter_size_surf" and "filter_size_map" help? Currently, they are set as 0.5.
3. What do you meant by degeneration? Could enlighten more?

Thanks for your great helps.

closed 2023-01-01 by stale[bot]

Shamiikebob, 2025-01-24:
Same issue here. It seems that FASTLIO produces this issue when the data gathered is too large for the data buffer. I am using a 3D which means the number of points will always be large. Yet, I still get this isssue.

XXLiu-HNU, 2025-05-30:
I also encountered this problem, but it didn't happen often. Sometimes it would happen 2-3 times in a day's experiment, and restarting the radar driver and algorithm would restore it to normal.