hku-mars/FAST_LIO #70: IMU  offset on avia_indoor_quick_shake_example1 and avia_indoor_quick_shake_example2

Camilochiang, 2021-09-09:
Hei.

In both of your example avia_indoor_quick_shake_example1 and avia_indoor_quick_shake_example2, there is an offset in the linear_acceleration.z of 0.99 and -.98 respectively. When this offset is corrected, FAST_LIO do not converge and loose track of the position, so the point cloud do not match.  See below

[image]

1. Do you know if this offset was set it by default (either 1 or -1) by Livox?
2.  Do you know if livox is pre-processing/filtering the IMU values before publishing it?
3.  I have an external BMI088 + mid70 and I'm correcting by the factory offset, so I'm afraid that FAST_LIO may have this problem with external IMUs.  So far I have only tried in a fix position and it works (as at the beginning of the bag example_1), but once that I start moving , I guess that I will have the same issue.
4. Also just to confirm, in which units is FAST_LIO imu data working: m s-1 and dgs  if im correct, no?

Greetings.

XW-HKU, 2021-09-09:
the imu unit in livox lidar is g，1g is around 9.8m /s2. FAST LIO automatically support different units. but It should be noted that imu is measuring the total acceleration except for the gravity. so when lidar is steady at a point, the imu will measure the support force which is the negative of gravity, that is exactly the offset 0.99 you mentioned.

Camilochiang, 2021-09-09:
Thanks for the quickly answer. 
What about the angular velocity ? Does FAST_LIO use rad s-1 as ROS recommend or dps?

Thanks!

XW-HKU, 2021-09-09:
angular velocity must be in unit rad / s.

Camilochiang, 2021-09-09:
Hei thanks!

Great. I will mark the topic as closed

closed 2021-09-09 by Camilochiang

mohsenhadadi, 2021-12-16:
Hi everyone I have a similar problem with loosing track ,
When I use Fast-Lio for Livox Horizon mounted at a 90 degrees perpendicular to my IMU's Y axis ,I encountered the following errors that I have attached. Please explain a little more about the parameters that I highlighted
, I only encounter such a problem when I change livox angle related to my IMU angle downwards among Y axis ,
anybody can help me solving this problem??
[image]
[image]
[image]

much appreciated,