hku-mars/FAST_LIO #71: crash after a long corridor in nclt dataset

HaisenbergPeng, 2021-09-09:
Dear Dr.Xu:
Thanks for sharing such impressive work!
Recently I have been testing FAST_LIO with NCLT [http://robots.engin.umich.edu/nclt/](url). Most of the time, it outputs accurate odometry. But occasionally the odometry will suddenly get lost. 
1. I only used a custum-cut period of the nclt20120108 dataset to test the performance of the code under sensor-degradation case, that is, long-corridor. So it is not a long-term issue;
2. The odometry lost happends usually indoor. So I changed 'blind' to 1 meter.
3. I disabled extrinsic_est_en and time_sync_en.
The dataset can be downloaded at [https://jbox.sjtu.edu.cn/l/XF3AoZ](url)

The velodyne.yaml is as below:
common:
    # lid_topic:  "/points_raw"
    lid_topic:  "/velodyne_points"
    imu_topic:  "/imu_6"
    time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
    
preprocess:
    lidar_type: 2                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line: 32
    blind: 1 # why blind?

mapping:
    acc_cov: 0.01608843085721201
    gyr_cov: 0.0017892995284273933
    b_acc_cov: 0.00035756256498712546
    b_gyr_cov: 1.5882455476866487e-05
    fov_degree:    180
    det_range:     100.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic
    # Lidar_T_wrt_IMU OR T that transforms imu coor. to lidar coor.?
    extrinsic_T: [ 0.17664, -0.10634, 0.24904]
    extrinsic_R: [ -0.012269, -0.99992, -0.0028972, 0.99983, -0.012309, 0.014084, -0.014119, -0.0027239, 0.9999]

publish:
    path_en:  false
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: false       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

XW-HKU, 2021-09-10:
the download link cannot access.

HaisenbergPeng, 2021-09-10:
Sorry, here is a google drive link [https://drive.google.com/file/d/1JlQhlzqNM8UBADmNRbIuw_yNW6lqzJkq/view?usp=sharing](url).

XW-HKU, 2021-09-10:
> Sorry, here is a google drive link [https://drive.google.com/file/d/1JlQhlzqNM8UBADmNRbIuw_yNW6lqzJkq/view?usp=sharing](url)

access require is sent, please check your email.

XW-HKU, 2021-09-10:
Looks like your data does not has
1. each lidar point' timestamp, the error message show _**Failed to find match for field 'time'.**_
2. each lidar's line number, the error message show **_Failed to find match for field 'ring'._**

HaisenbergPeng, 2021-09-11:
I tried your script with python 3.7(scipy 1.7.1, numpy 1.20 or 1.16.5) but got such error: 
TypeError: can't set attributes of built-in/extension type 'scipy.spatial.transform.rotation.Rotation'
Could you please share your version numbers of python, scipy and numpy? thanks!

XW-HKU, 2021-09-11:
> I tried your script with python 3.7(scipy 1.7.1, numpy 1.20 or 1.16.5) but got such error:
> TypeError: can't set attributes of built-in/extension type 'scipy.spatial.transform.rotation.Rotation'
> Could you please share your version numbers of python, scipy and numpy? thanks!

python 3.6.9, scipy 1.5.4, numpy 1.19.5.

XW-HKU, 2021-09-11:
I may updated the script several hours ago, please try again.

HaisenbergPeng, 2021-09-11:
Dr.Xu, I fixed he error by removing "R.r" in line 157 and line 189 (Honestly I don't know what "R.r" is for).
BTW, did you convert IMU data to lidar frame in the script？

HaisenbergPeng, 2021-09-11:
Dr.Xu, thanks a lot for your python script(I actually took a while trying to write such a script like yours, but didn't make it as good). With the generated bag file of nclt20120108, I was able to get very neat result.
Still, I have two questions: 
1. After I deleting R.r from the script, I still can get the attitude measurement in the bag file. I don't see the necessity of R.r.
(            # r_q = R.r = R.from_euler('xyz', [h, p, r], degrees=0)
            r_q = R.from_euler('xyz', [h, p, r], degrees=0) ) 
2. Did you get rid of intensity data conversion in the bag file just to save the storage space of the bag file?

XW-HKU, 2021-09-11:
1. that quaternion is for some lidar slam, such as lios-sam, needs 9-axis imu （9-axis imu can produce quaternion）. 
2. there is a bug in **pcl2.create_pointcloud()** finction: 

if you build **time** field in format **FLOAT32**, you cannot build **intensity** field at the same time. i dont know how to fix it so i didn't generate the **intensity** field.

You may could find a better pointcloud2 generating method to replace it.

HaisenbergPeng, 2021-09-12:
Hi, with a minor revision on your script, It seems that I shoved the intensity into the bag file, please take a look at [the script](https://drive.google.com/file/d/1Ssca_j25dQLUz1nDZSPIEowb6Fzh9k9q/view?usp=sharing).
[image]

XW-HKU, 2021-09-12:
> Hi, with a minor revision on your script, It seems that I shoved the intensity into the bag file, please take a look at [the script](https://drive.google.com/file/d/1Ssca_j25dQLUz1nDZSPIEowb6Fzh9k9q/view?usp=sharing).
> [image]

That's great! Thank you very much!

closed 2021-12-24 by HaisenbergPeng