hku-mars/FAST_LIO #72: 请问extrinsic参数

qianxingdu, 2021-09-10:
您好：
在avia.yaml文件中，有几个问题想请问一下，十分感谢！
（1）extrinsic_T的单位是米吗？
（2）对于extrinsic_R，只要IMU和激光雷达的xyz轴是一样的，该参数是不是就不需要修改。
（3）对于extrinsic_est_en，在线标定的原理是什么呢，如果将extrinsic_est_en设置为true，那么extrinsic_T参数需要置为[0, 0, 0]吗？我实际尝试了下将extrinsic_est_en设置为true，但是发现效果并不好，不知道是什么原因。

XW-HKU, 2021-09-10:
1. 是米，国际单位制。
2. 如果xyz对齐了，那R就是默认单位制。
3. 这个在线估计需要初值不能差距过大，不然没法收敛。

效果不好的话建议检查雷达帧内点的时间是否正确，尤其是单位。

qianxingqx, 2021-09-13:
非常感谢您的解答！
我后面又调整了一下，发现将 avia.yaml 文件中的 gyr_cov 置为 0 的时候，效果还可以，所以还想请问一下 acc_cov, gyr_cov, b_acc_cov, b_gyr_cov 的初值是如何得到的呢？
还有就是 fov_degree 和 det_range 参数是只针对 AVIA 雷达吗，我用的 Mid 40，需要修改这两个参数吗？
谢谢您的耐心解答！

XW-HKU, 2021-09-13:
gyr_cov不应该设置为0。

有两个问题必须先确认，否则fastlio无法工作：
1. imu和mid40之间有没有硬件时间戳同步。
2. mid40是否有每个雷达点的time，也就是运行的时候有没有黄色字体报错。

另外，提issue请一开始就提供雷达型号，imu型号，PC型号，场景情况。这些信息非常重要！没有这些信息，没法判断什么原因。

XW-HKU, 2021-09-13:
> 非常感谢您的解答！
> 我后面又调整了一下，发现将 avia.yaml 文件中的 gyr_cov 置为 0 的时候，效果还可以，所以还想请问一下 acc_cov, gyr_cov, b_acc_cov, b_gyr_cov 的初值是如何得到的呢？
> 还有就是 fov_degree 和 det_range 参数是只针对 AVIA 雷达吗，我用的 Mid 40，需要修改这两个参数吗？
> 谢谢您的耐心解答！

我没有在mid40上做过任何测试。

qianxingqx, 2021-09-13:
不好意思，之前有所疏忽！
我的雷达型号是：Livox Mid 40，imu 型号：超核Hi226，PC：Ubuntu 16.04，场景：在室内进行扫描。

1. 由于目前没有硬件同步的条件，所以我修改了 livox_ros_driver 中的程序，通过 ros::Time::now() 获取当前时间戳来发布 lidar_msg 的点云，imu 的时间戳也是 ros::Time::now()。相当于使用 ROS 进行软同步，通过输出发现 lidar 和 imu 的时间相差约 0.1 秒，不过由于消息发送也有延迟，所以这种同步可能也不够准确。
2. Mid 40 运行的时候没有黄色字体报错，我通过 rostopic echo 查看了一下，是有每个点的时间戳的，如下：
`
  - 
    offset_time: 99996550
    x: 2.78399991989
    y: -0.428999990225
    z: -0.741999983788
    reflectivity: 24
    tag: 0
    line: 0
  - 
    offset_time: 100006550
    x: 2.78099989891
    y: -0.428000003099
    z: -0.737999975681
    reflectivity: 25
    tag: 0
    line: 0
  - 
    offset_time: 100016550
    x: 2.78500008583
    y: -0.428000003099
    z: -0.735000014305
    reflectivity: 29
    tag: 0
    line: 0
`

XW-HKU, 2021-09-13:
> 不好意思，之前有所疏忽！
> 我的雷达型号是：Livox Mid 40，imu 型号：超核Hi226，PC：Ubuntu 16.04，场景：在室内进行扫描。
>
> 1. 由于目前没有硬件同步的条件，所以我修改了 livox_ros_driver 中的程序，通过 ros::Time::now() 获取当前时间戳来发布 lidar_msg 的点云，imu 的时间戳也是 [...]

1. 这种软同步什么都保证不了，因为传感器和ros之间的通信延迟你根本就无从得知。
2. offset time单位看上去是对的，但是由于没做时间同步，去畸变很可能都是错的，精度无法保证。
3. 室内场景是什么样的，给照片或者具体描述。3米内很小的场景fastlio不支持。
4. imu和雷达之间的外参设置正确吗

qianxingqx, 2021-09-13:
非常感谢您的耐心回复！
1. 软同步确实不够准确，后面我尝试一下使用硬件同步好了
2. 室内场景：在一个房间中环绕一个约2米高的桌子一周进行扫描，扫描距离有时候可能是小于3米的。我之前也尝试过 LOAM，效果也不太好，请问下那这种小场景的话有适合的方法吗？
3. imu 和雷达之间的外参设置是手动用直尺进行测量的
顺便想请问一下您之前说的 `gyr_cov不应该设置为0` 是为什么呢？

XW-HKU, 2021-09-13:
> 非常感谢您的耐心回复！
> 
> 1. 软同步确实不够准确，后面我尝试一下使用硬件同步好了
> 2. 室内场景：在一个房间中环绕一个约2米高的桌子一周进行扫描，扫描距离有时候可能是小于3米的。我之前也尝试过 LOAM，效果也不太好，请问下那这种小场景的话有适合的方法吗？
> 3. imu 和雷达之间的外参设置是手动用直尺进行测量的
>    顺便想请问一下您之前说的 `gyr_cov不应该设置为0` 是为什么呢？

你说的效果不好到底是什么样的效果不好？能不能把信息提供全一点啊？

qianxingqx, 2021-09-13:
例如，在以下场景中：
[image]
我从右往左顺时针绕着中间的物体扫描，放大如下图
[image]
可以kanchu看出其边缘是比较模糊的，同一平面下的点云没有很好地配准，而且上面两个图是在 gyr_cov 参数设置为 0 的情况下运行出来的。

如果将 gyr_cov 设置为您之前的 0.1，那么位姿就会漂移，如下图：
[image]
可以看出在扫描到一半的时候，位姿就快速地向右漂移了

XW-HKU, 2021-09-13:
这个场景都是退化场景，fastlio几乎不可能work.

qianxingqx, 2021-09-13:
请问这种退化是由什么原因导致的呢

XW-HKU, 2021-09-13:
雷达slam需要至少三个不平行的平面才能进行定位。这种室内的，还是mid40这种超级小fov lidar，很难保证不退化。退化只是问题之一，我觉得时间同步你需要优先解决。

检查时间同步或者imu外参是否准确的一个简单方法，观察一帧雷达数据中的平面在雷达运动的时候是否能保持平整，如果一帧内的平面分层了，那么说明数据本身（时间戳同步，imu外参）存在问题。

qianxingqx, 2021-09-13:
好的，非常感谢您的耐心解答！后面我会尝试先解决时间同步的问题。

closed 2021-09-22 by qianxingqx

Doodle1106, 2021-12-29:
@XW-HKU 你好，
你好，我跑过一个1000x1000的场景，走的多个U turn， 建图结果很好，平面没有重影，轨迹平滑，我开启了在线标定，同时给了一个外参yaw上面90度的offset， 但是在线估计出来的结果并不理想，请问FAST LIO在建图良好的情况下外参估计不准的问题是因为平面运动导致的吗？