hku-mars/FAST_LIO #74: Large drift on utbm dataset

spc2, 2021-09-14:
Thanks for sharing your great work!
Description:
I have tested FAST_LIO on utbm[https://github.com/epan-utbm/utbm_robocar_dataset](url) dataset recently. Most of time, it produces an accurate odometry. But on the utbm_2019_04_18(Roundabout) data, it suffers a large drift.
[image]

My config file velodyne.yaml is as below:
common:
    lid_topic:  "/velodyne_points"
    imu_topic:  "/imu/data"
    time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
    
preprocess:
    lidar_type: 2                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line: 32
    blind: 4

mapping:
    acc_cov: 0.1
    gyr_cov: 0.1
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    fov_degree:    180
    det_range:     100.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic
    extrinsic_T: [ -0.5, 1.4, 1.5]
    extrinsic_R: [ 1, 0, 0, 
                         0, 1, 0, 
                         0, 0, 1]

publish:
    path_en:  false
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

XW-HKU, 2021-09-14:
did you meet the yellow warning: "Failed to find match for field 'time' "?

XW-HKU, 2021-09-14:
Did you try extrinsic_T: [ -0.5, 1.4, 1.5]?

spc2, 2021-09-14:
Thank you for your responses!
When I run FAST_LIO on the utbm_20190418_roundabout data, everything is normal.  There is no yellow warning and the extrinsic_T just sets [ -0.5, 1.4, 1.5].

XW-HKU, 2021-09-15:
What will happens if you set the try extrinsic_T to [ 0, 0, 0.28]?

spc2, 2021-09-16:
It also suffers a large drift just like using [ -0.5, 1.4, 1.5].

closed 2023-01-01 by stale[bot]