hku-mars/FAST_LIO #82: Testing mid 70

Camilochiang, 2021-09-30:
Im testing your code with a MID 70 + external IMU (BMI088, same orientation and similar location as in AVIA).
In some cases the algorithm just drift appart, as for example in the attached bag after aprox 20 seconds of good matching. 
Any idea why this could be happening?
[image]

Ps: you can find the ROSBAG and configuration files [here](https://drive.google.com/drive/folders/1wVugljpp0x9cHzMQdfczS47NgpD5DiXl?usp=sharing):

Camilochiang, 2021-09-30:
[image]

XW-HKU, 2021-09-30:
The environment seems to be a degeneration scene.

Camilochiang, 2021-09-30:
What conditions/environments may generate this ? : Too many objects close to the LIDAR ? to little ones?

XW-HKU, 2021-09-30:
If LiDAR only watches one or two large surfaces (for example only face
floor or ground), LiDAR cannot locate itself.

Camilochiang, 2021-09-30:
OK! That means that the algorithm do not work in open spaces so well . I guess that this problem should not happen with R2Live as the camera may help no?

Thank in any case!

XW-HKU, 2021-09-30:
It is not about open space. The LiDAR cannot be put only facing the wall or ground.

Of course camera will help.

closed 2021-09-30 by Camilochiang

maomao0220, 2024-06-01:
在程序中，哪部分的代码会标志着程序“飞”了？