hku-mars/FAST_LIO #83: Could not find a package configuration file provided by "pcl_ros"

CogmanCVN, 2021-10-04:
Hello， when executing the command **_catkin_ make_**, I have this problem:

CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "pcl_ros" with any
  of the following names:

    pcl_rosConfig.cmake
    pcl_ros-config.cmake

  Add the installation prefix of "pcl_ros" to CMAKE_PREFIX_PATH or set
  "pcl_ros_DIR" to a directory containing one of the above files.  If
  "pcl_ros" provides a separate development package or SDK, be sure it has
  been installed.

CogmanCVN, 2021-10-04:
Ubuntu 18.04 execute the following command to solve the problem:
sudo apt-get install ros-melodic-pcl-ros

XW-HKU, 2021-10-10:
exactly

> Ubuntu 18.04 execute the following command to solve the problem: sudo apt-get install ros-melodic-pcl-ros

closed 2021-10-10 by XW-HKU