hku-mars/FAST_LIO #94: The difference of new df_dx and df_dw matrix after using ikfom

daijicheng, 2021-10-31:
Thanks for your brilliant work. 
After switching to ikfom, the function _df_dx_ and _df_dw_ in use-ikfom.hpp **line 52** and **line 71** have changed. However, due to the lack of relevant information about the function _predict_ of ikfom, the current changes are difficult to understand. Is it convenient for the author to give a description of the new matrix.

Joanna-HE, 2021-10-31:
In the use-ikfom.hpp, there is also a f function that is defined according to the system. df_dx and df_dw are calculated by differentiating the f function with respect the state. The reason for calculating in this way is that the ikfom give a new formulation to the system, which makes it easy to deploy an iterated EKF for a certain system.

ZhangZh3ng, 2022-11-16:
> In the use-ikfom.hpp, there is also a f function that is defined according to the system. df_dx and df_dw are calculated by differentiating the f function with respect the state. The reason for [...]

请问，您能更详细的解释一下df_dx的计算原理吗，代码采用的算法已经与fast-lio论文中的公式7对不上了。

Joanna-HE, 2022-11-17:
df_dx的计算就是f函数对x的导数，其中f函数和fast-lio论文中定义的一样，区别在于，df_dx不是最终展示在fast-lio论文中的雅可比矩阵的结果，还需要和ikfom中自行计算的与manifold相关的结果结合起来，才是最终的雅可比矩阵。
________________________________
寄件者: Zhang Zheng ***@***.***>
寄件日期: 2022年11月16日 下午 03:14
收件者: hku-mars/FAST_LIO ***@***.***>
副本: He Dongjiao ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/FAST_LIO] The difference of new df_dx and df_dw matrix after using ikfom (Issue #94)

In the use-ikfom.hpp, there is also a f function that is defined according to the system. df_dx and df_dw are calculated by differentiating the f function with respect the state. The reason for calculating in this way is that the ikfom give a new formulation to the system, which makes it easy to deploy an iterated EKF for a certain system.

请问，您能更详细的解释一下df_dx的计算原理吗，代码采用的算法已经与fast-lio论文中的公式7对不上了。

closed 2023-01-01 by stale[bot]