hku-mars/FAST_LIO #98: downsampled vs raw points

patripfr, 2021-11-17:
Hi,
thanks for a lot for providing this repo.
In your paper about FAST-LIO2 you state in several places that you are aligning the raw points against the map. However when I was looking through your code, it seems to me that you are actually downsampling the point cloud using a voxel filter, and then align this downsampled pointcloud, which for my understanding does not contain raw points anymore.

https://github.com/hku-mars/FAST_LIO/blob/d0a877db69f569964697e1135e1fcb42aae9b3f2/src/laserMapping.cpp#L892-L894

Am I missing something here or could you give me some insights on this discrepancy?
Thanks!

XW-HKU, 2021-11-17:
The direct method means not using feature points (including plane feature and edge features). The voxel filter used in fastlio2 does not generate any new lidar points, but keeps the raw point closest to the voxel centers. Also fast lio 2 never states that we use "all" raw points, we only claims that we removes the feature extraction modules and directly register the raw points.

patripfr, 2021-11-17:
>  The voxel filter used in fastlio2 does not generate any new lidar points, but keeps the raw point closest to the voxel centers.
I don't think that this statement is correct, in the lines I cited above you are using the standard PCL voxelgrid filter, which does not keep the raw point closest to the voxel center, but instead calculates the centroid of the points in each voxel.
Anyways, thanks for pointing this out.

closed 2021-11-17 by patripfr