hku-mars/FAST_LIO #100: Multiple Lidars support

raspectedward, 2021-11-24:
May we know if this supports multiple Horizon Lidars? Thanks

AHuguetUPC, 2021-11-26:
The algorithm itself does not care where does the pointcloud come from, so it would be pretty direct to do!

Have you tried adding an additional subscriber calling the same callback, given the timestamps correctly I guess it should work. It's actually an interesting try...

Andreu

raspectedward, 2021-11-30:
so do you mean I just give out the combined rosbag from multiple Horizon lidars. The algorithm should work?

XW-HKU, 2021-11-30:
> so do you mean I just give out the combined rosbag from multiple Horizon lidars. The algorithm should work?

if you make sure the extrinsics and timestamps are all right，that should work.

raspectedward, 2021-12-01:
we tried FAST LIO with multiple lidars. It first generates the integrated map then change to pcd. The map generation part has errors when more than one lidar.

closed 2023-01-01 by stale[bot]