hku-mars/FAST_LIO #101: Velodyne 十六线雷达

huduo0812, 2021-11-24:
你好，感谢你们开源这么好的作品。
我使用livox_avia构建的地图效果非常好，但是有时候会z轴漂移，而且数据量太大，生成的地图无法导航，
现在我打算使用Velodyne的雷达构建地图，根据提示配置完成后，一开始运行时，点云图就会旋转，然后不久就直接飞走了，请教一下可能是什么问题。谢谢！！

AHuguetUPC, 2021-11-26:
Hello!
It seems maybe your IMUs and Velodyne axis are not aligned... Have you checked? Change the R variable in the YAML file.

If it's not that (99% it is!), send a video of what happens.

Andreu

huduo0812, 2021-11-26:
> Hello! It seems maybe your IMUs and Velodyne axis are not aligned... Have you checked? Change the R variable in the YAML file.
> 
> If it's not that (99% it is!), send a video of what happens.
> 
> Andreu

谢谢，应该是你说的这个问题，我把imu和Velodyne 按照坐标系放在同一个方向标定过它们的旋转矩阵，然后修改ymal里面的extrinsic_T和extrinsic_R参数，但是也是飞走了。可能是标定的结果有问题，请问你是怎么标定的这两个参数？

Huguet57, 2021-11-28:
By "calibrating in the same coordinate system" I guess you mean putting -1 and changing columns of the R matrix. Have you tried the algorithm with these rosbags they give as example: [Drive with Velodyne Rosbags](https://drive.google.com/drive/folders/1VBK5idI1oyW0GC_I_Hxh63aqam3nocNK ), [Section in README](https://github.com/hku-mars/FAST_LIO#42-velodyne-hdl-32e-rosbag).

If those work, as they should, try to compare the IMU data and Velodyne pointclouds of the rosbags to yours. Try to make transformations (change of axis, basically) of your data to replicate the same axis as the example data.

On another hand, what's your IMU frequency? If it is too low it can be problematic, but check the axis thing.

Andreu

huduo0812, 2021-11-29:
> 通过“在同一坐标系中校准”，我猜您的意思是放置-1并更改- [R矩阵的列。您是否尝试过使用这些rosbags作为示例的算法：[驱动器，带Velodyne Rosbags](https://drive.google.com/drive/folders/1VBK5idI1oyW0GC_I_Hxh63aqam3nocNK)，[自述文件部分](https://github.com/hku-mars/FAST_LIO#42-velodyne-hdl-32e-rosbag)。
> 
> 如果这些工作正常，请尝试将 rosbags 的 IMU 数据和 Velodyne 点云与您的数据进行比较。实验对数据进行转换（几乎是的轴变化）以复制与示例数据相同的轴。
> 
> 相反，你的 IMU 频率是多少？如果它太低，可能会有问题，但请检查轴的东西。
> 
> 安德烈

你好，我没用试过你说的这些，对于我来说有点复杂，我现在是用livox_avia录个数据包，然后在工作站(高性能电脑)上运行这个建图算法，然后用Velodyne 雷达在建好的pcd图中进行导航定位，效果还不错！！！

Huguet57, 2021-11-29:
Mmh... So I have understood correctly, you want to use the Velodyne LiDAR to build a map and then use a Livox Avia to navigate on it. And that's what's failing.

Can you send a video of the effect that happens? It's an interesting question because it should work in principle.

Andreu

huduo0812, 2021-11-29:
> 嗯...所以我理解正确，您想使用 Velodyne LiDAR 构建地图，然后使用 Livox Avia 在其上导航。这就是失败的原因。
> 
> 你能发一个发生的效果的视频吗？这是一个有趣的问题，因为它在原则上应该有效。
> 
> 安德烈

哈哈哈哈，你理解反了，是Livox Avia构建地图，Velodyne 进行导航定位，给你看一下下午导航的效果. 

https://user-images.githubusercontent.com/81087363/143873722-2813bc74-39e7-4f52-9ec6-b9382be50a55.mp4

https://user-images.githubusercontent.com/81087363/143873738-458b7bb8-a5cd-429a-86eb-cda18aed9857.mp4

XW-HKU, 2021-11-29:
检查一下imu测量，静止的时候，imu的测量应当是负的重力加速度而不是0，0， 0

huduo0812, 2021-11-29:
> 检查一下imu测量，静止的时候，imu的测量应当是负的重力加速度而不是0，0， 0

这是我们imu的加速度数据，z轴是正的

https://user-images.githubusercontent.com/81087363/143874861-a856a733-5341-49f4-8ca0-bc9ed97a0804.mp4

，

Huguet57, 2021-11-29:
I meant a video of RViz 😭😂

On a second thought: shouldn't the IMU measurement be (0,0,-9.81...) instead of (0,0,1)?

XW-HKU, 2021-11-29:
> > 检查一下imu测量，静止的时候，imu的测量应当是负的重力加速度而不是0，0， 0
> 
> 这是我们imu的加速度数据，z轴是正的
> 
>  b0af197ef3214239edb76bfcb2cc5e77.mp4 
> ，

你的雷达的z轴和imu的z轴是否都是朝上的？

huduo0812, 2021-11-29:
> 我是说 RViz 的视频 😭😂
> 
> 再想一想：IMU 测量值不应该是 (0,0,-9.81...) 而不是 (0,0,1)？

rviz里面你想看哪些东西，明天我有考试，后天录个视频给你看😂，  imu那个我也不知道怎么回事。。。。

huduo0812, 2021-11-29:
> > > 检查一下imu测量，静止的时候，mu的测量应该是负的而不是0，0，00
> > 
> > 
> > 这是我们imu的正数据，z轴是的
> > b0af197ef3214239edb76bfcb2cc5e77.mp4 
> > ，
> 
> 你的雷达的z轴和imu的z轴是否都是朝上的？

[image]

[image]
我们是这样装的，都是朝上。

XW-HKU, 2021-11-29:
外接的imu要做好时间同步

huduo0812, 2021-11-29:
> 外接的imu要做好时间同步

这个时间同步太难了，现在我们用Livox Avia 进行建图，它有内置的imu，然后在将点云图向下采样，变得稀疏一点。Velodyne定位导航，之前的建图算法z轴漂移问题比较严重，地图像帆船一样，两边都翘起来了，现在用你这个算法建图的效果非常好，哦，对了，顺便问一下，这个建图算法有回环检测吗，我们是在室外录的数据包，然后回来用工作站运行建图算法，有时候转弯太快就会整体漂移。

XW-HKU, 2021-11-29:
readme里面有链接到loop closure的项目。

huduo0812, 2021-11-30:
> readme里面有链接到loop closure的项目。

好的，谢谢，我去看看。！

huduo0812, 2021-12-01:
> readme里面有链接到loop closure的项目。

你好，我又来了，这次碰到新的问题了，这次我用avia录了个22G大小的包，生成的scans.pcd有9.8G大，但是用pcl_viewer查看这个点云会报段错误，然后用其他程序打开也是段错误，这是为什么呢？谢谢你的帮忙！

huduo0812, 2021-12-01:
> > readme里面有链接到loop closure的项目。
> 
> 你好，我又来了，这次碰到新的问题了，这次我用avia录了个22G大小的包，生成的scans.pcd有9.8G大，但是用pcl_viewer查看这个点云会报段错误，然后用其他程序打开也是段错误，这是为什么呢？谢谢你的帮忙！

我发现生成的pcd文件大于4个G的话，pcl就无法读取了。

XW-HKU, 2021-12-01:
是你内存不够了吧

huduo0812, 2021-12-01:
> 是你内存不够了吧

不应该吧，内存256G呢。。。我明天试试更高版本的pcl，我现在用的是1.8的
[image]

closed 2023-01-01 by stale[bot]

charon-cheung, 2023-10-11:
> > 是你内存不够了吧
> 
> 不应该吧，内存256G呢。。。我明天试试更高版本的pcl，我现在用的是1.8的 [image]

你们是怎么进行定位的？导航又是怎么做的？难道转为栅格地图再导航？

huduo0812, 2023-10-11:
> > > 是你内存不够了吧
> > 
> > 
> > 不应该吧，内存256G呢。。。我明天试试更高版本的pcl，我现在用的是1.8的 [image]
> 
> 你们是怎么进行定位的？导航又是怎么做的？难道转为栅格地图再导航？

时间太久了，我记得是3d定位，2d导航，最终就是栅格地图。

charon-cheung, 2023-11-02:
> commented
你现在还熟悉fast_lio吗，能不能加个微信交流一下

huduo0812, 2023-11-02:
> > commented
> > 你现在还熟悉fast_lio吗，能不能加个微信交流一下

不好意思，我现在在学深度学习，这个项目时间太久了i😢